openwrt/packages #4934: mii-tool: can't compile with kernel 4.9.x

dibdot, 2017-10-12:
@Borromini 

can't compile mii-tool with kernel 4.9.x (target ar71xx):

```
make[5]: Leaving directory '/home/dirk/lede/GL-MIFI/source/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f/man'
mips-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/dirk/lede/GL-MIFI/source/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f:mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -I/home/dirk/lede/GL-MIFI/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/dirk/lede/GL-MIFI/source/staging_dir/target-mips_24kc_musl/include -I/home/dirk/lede/GL-MIFI/source/staging_dir/toolchain-mips_24kc_gcc-6.3.0_musl/usr/include -I/home/dirk/lede/GL-MIFI/source/staging_dir/toolchain-mips_24kc_gcc-6.3.0_musl/include/fortify -I/home/dirk/lede/GL-MIFI/source/staging_dir/toolchain-mips_24kc_gcc-6.3.0_musl/include  -Wall -fno-strict-aliasing  -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -I. -I/home/dirk/lede/GL-MIFI/source/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f/include -Ilib -c iptunnel.c
In file included from /home/dirk/lede/GL-MIFI/source/staging_dir/toolchain-mips_24kc_gcc-6.3.0_musl/include/linux/if_tunnel.h:6:0,
                 from iptunnel.c:34:
/home/dirk/lede/GL-MIFI/source/staging_dir/toolchain-mips_24kc_gcc-6.3.0_musl/include/linux/ip.h:85:8: error: redefinition of 'struct iphdr'
 struct iphdr {
        ^~~~~
In file included from iptunnel.c:29:0:
/home/dirk/lede/GL-MIFI/source/staging_dir/toolchain-mips_24kc_gcc-6.3.0_musl/include/netinet/ip.h:25:8: note: originally defined here
 struct iphdr {
        ^~~~~
Makefile:106: recipe for target 'iptunnel.o' failed
make[4]: *** [iptunnel.o] Error 1
make[4]: Leaving directory '/home/dirk/lede/GL-MIFI/source/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f'
Makefile:54: recipe for target '/home/dirk/lede/GL-MIFI/source/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f/.built' failed
make[3]: *** [/home/dirk/lede/GL-MIFI/source/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f/.built] Error 2
make[3]: Leaving directory '/home/dirk/lede/GL-MIFI/source/feeds/packages/net/mii-tool'
package/Makefile:109: recipe for target 'package/feeds/packages/mii-tool/compile' failed
make[2]: *** [package/feeds/packages/mii-tool/compile] Error 2
make[2]: Leaving directory '/home/dirk/lede/GL-MIFI/source'
package/Makefile:105: recipe for target '/home/dirk/lede/GL-MIFI/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/dirk/lede/GL-MIFI/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dirk/lede/GL-MIFI/source'
/home/dirk/lede/GL-MIFI/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
```
if I change iptunnel.c like that, compile runs fine:

```
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <syslog.h>
#include <fcntl.h>
#include <sys/socket.h>
#include <sys/ioctl.h>
#include <netinet/in.h>
/* #include <netinet/ip.h> */
#include <arpa/inet.h>
#include <net/if.h>
#include <net/if_arp.h>
#include <linux/types.h>
#include <linux/ip.h> /* new */
#include <linux/if_tunnel.h>
```

Borromini, 2017-10-18:
@dibdot does this work on other architectures as well?

I haven't built trunk for a while, won't be able to test myself until next week.

dibdot, 2017-10-19:
@Borromini 
yes, I've tested the header change on x86 as well, works fine with kernel 4.9.x.

Borromini, 2017-10-19:
@dibdot thanks for checking. Can't we remove that line altogether then, if it breaks things? Or would you rather keep it commented for future reference?

dibdot, 2017-10-19:
@Borromini 
the proper fix is even simpler ...  ;-)

iptunnel.c includes netinet/ip.h which contains a definition of the iphdr struct.
iptunnel.c also includes linux/if_tunnel.h which includes linux/ip.h which contains a definition of the iphdr struct. So, both netinet/ip.h and linux/ip.h define the iphdr struct, and both of them have been included directly or indirectly by iptunnel.c. Because of that the compilation fails due to a struct redefinition.

The problem can be solved by just not include netinet/ip.h (tested on ar71xx & x86).

Borromini, 2017-10-22:
@dibdot 

I just tested without netinet/ip.h myself, and I'm seeing some weird stuff. Building went fine on ramips and x86/64 but on ar71xx it breaks for me:

`/var/data/code/lede/main/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/include/linux/if_tunnel.h:37:16: error: field 'iph' has incomplete type
  struct iphdr  iph;
                ^
Makefile:106: recipe for target 'iptunnel.o' failed
make[3]: *** [iptunnel.o] Error 1
make[3]: Leaving directory '/var/data/code/lede/main/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f'
Makefile:54: recipe for target '/var/data/code/lede/main/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f/.built' failed
make[2]: *** [/var/data/code/lede/main/build_dir/target-mips_24kc_musl/mii-tool-2016-10-06-115f1af2494ded1fcd21c8419d5e289bc4df380f/.built] Error 2
make[2]: Leaving directory '/var/data/code/lede/main/feeds/packages/net/mii-tool'
package/Makefile:109: recipe for target 'package/feeds/packages/mii-tool/compile' failed`

This is on a fresh master toolchain (from a few days ago), on a Debian Stretch install. 

If I leave iptunnel.c untouched it will build on ar71xx somehow but not on x86/64 or ramips. Will wipe my toolchain again, start afresh and report back.

dibdot, 2017-10-22:
That will happen, if you build against kernel 4.4.x toolchain. Since today the default for ar71xx has been changed to 4.9.x (see https://git.lede-project.org/?p=source.git;a=commit;h=c84b7eaedd1b3c9b7e3e30248ab3d0050477f6d5)

Borromini, 2017-10-24:
Silly me, I saw the diffconfig indicating the switch from 4.4 to 4.9. It took a new checkout somehow to get ar71xx to build with 4.9 and the removed include, but I've opened a PR. Thanks for doing the legwork!

dibdot, 2017-10-29:
@Borromini Thank you very much!

closed 2017-10-29 by dibdot