openwrt/packages #4941: <package-name>: Squid - cache_dir type 'rock'

FangsMcWolf, 2017-10-15:
When using the cache_dir directive and specifying the storage type as 'rock,' an issue occurs when trying to use the 'squid -z' in the command line.

```2017/10/15 05:48:53| assertion failed: ../src/ipc/AtomicWord.h:88: "Enabled()"```

A temporary work-around is to remove the directive, have Squid rebuild the directory structure, then re-add the directive to the conf file.  I can only assume that it's working beyond that, as it doesn't report any other errors (such as it not being a supported type).

diogosimao, 2018-03-22:
Hi there!
`cache_dir` in my config is specified as:
`cache_dir aufs /tmp/squid/cache 60000 16 512`

Would you say that squid should accept following setup in order to work gracefully?

```
root@LEDE:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.3M      2.3M         0 100% /rom
tmpfs                    13.8M      1.9M     11.9M  14% /tmp
/dev/sda1                10.9G     60.7M     10.3G   1% /overlay
overlayfs:/overlay       10.9G     60.7M     10.3G   1% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/sda3               811.2G    590.8G    179.1G  77% /mnt/sda3
/dev/sda4                87.7G     90.1M     83.2G   0% /tmp/squid
```
I followed [instructions](https://openwrt.org/docs/guide-user/services/proxy/proxy.squid) for transparent mode. I can't say if I have a device with or without enough flash space, so I ignored the second part of instructions. I know that I have plenty of space in my external storage and for packages and dependencies installations.

I am experience major router slow down and pages won't load after starting squid e apply firewall rule.

diogosimao, 2018-03-22:
Look like storage space is not the only necessary thing. It seens that my CPU, FLA, RAM are also bottlenecks, of course. Just compared a [ClearFog](https://wiki.openwrt.org/toh/solidrun/clearfog) with my [TP-LINK](https://wikidevi.com/wiki/TP-LINK_TL-WR1043ND_v1.x)

neheb, 2020-01-12:
Is this still an issue?

commodo, 2023-05-27:
closing;

feel free to re-open if i'm wrong

closed 2023-05-27 by commodo