openwrt/packages #4944: dhcpcd: needs a new maintainer

rsmarples, 2017-10-15:
I no longer use OpenWRT, nor have an environment to compile packages for it.
Please [remove me from the Makefile](https://github.com/openwrt/packages/blob/4006865ae81b20b1793ae2a07db20235fefd2c71/net/dhcpcd/Makefile#L22)

Happy to fix any reported issues though, I just can't compile or test it on the OpenWRT platform (or derivates).

ratkaj, 2017-10-19:
If there are no objections i might as well take it. I'll throw in an update tomorrow, seems to be  outdated.

diizzyy, 2017-12-12:
@rsmarples 
I think this is sorted now, can you close this issue?
Thanks for your contribution and future ones! :-)

rsmarples, 2017-12-12:
Happy to close it, thanks for taking dhcpcd over in OpenWRT @ratkaj

closed 2017-12-12 by rsmarples