openwrt/packages #4946: ddns-scripts: unable to use proxy with https request

sequencer, 2017-10-16:
Maintainer: @chris5560 
Environment: NETGEAR WNDR3700v4 | LEDE Reboot 17.01.2 r3435-65eec8bd5f / LuCI lede-17.01 branch (git-17.152.82987-7f6fc16)
cause I'm in China, the google-domains are blocked from gfw.
I have a http proxy server in 10.10.0.120:6152, only in https & proxy method can my ddns be updated.
this is my conf:
```

config ddns 'global'
        option ddns_dateformat '%F %R'
        option ddns_loglines '250'
        option upd_privateip '0'

config service 'myddns_ipv4'
        option interface 'wan'
        option ip_source 'network'
        option ip_network 'wan'
        option service_name 'google.com'
        option proxy '10.10.0.120:6153'
        option lookup_host 'domains.google.com'
        option domain 'my.domain.com'
        option username 'username'
        option password 'passowrd'
        option enabled '1'
        option use_https '1'
```

if I don't use 'use_https' in the config, my proxy will receive the request. If I don't, my proxy will not receive the request.

Form the [Stackoverflow](https://stackoverflow.com/a/516398) the https over http proxy server method is provided.

chris5560, 2017-10-16:
From my understanding, reading the Stackoverflow article, https-traffic over http-proxy require a very special configuration.
But you can try to modify the code of dynamic_dns_updater.sh
search for "export HTTP_PROXY" (original code on line 314-317)
and replace the "http://$proxy" with "https://$proxy"
replace it on all 4 export lines, minimum the 2 lines for HTTPS_PROXY.
Don't know it it helps but no other idea.
wget/curl should use this exported environment.

sequencer, 2017-10-18:
it might be a bad idea.
https proxy is different from http proxy
as far as I know, maybe the Stackoverflow answer is a generally method for this situation.
for example in curl:
```
➜  ~ curl -x http://proxyserver:port https://www.google.com
<!doctype html><html itemscope="" itemtype="http://schema.org/WebPage" lang="en"><head><meta content="Search the world's information, including webpages, images, videos and more. Google has many special features to help you find exactly what you're looking for." name="description"><meta content="noodp" name="robots"><meta content="text/html; charset=UTF-8" http-equiv="Content-Type"><meta content="/images/branding/googleg/1x/googleg_standard_color_128dp.png" itemprop="image"><title>Google</title><script>(function(){window.google={kEI:'tJjnWbLaJcuV0gKZpJzoDw',kEXPI:'1352823,1353383,1353606,1354277,1354401,1354916,1355217,1355325,1355526,1355735,1355801,1355820,1356069,3700212,3700249,3700429,3700439,3700440,3700476,4029815,4031109,4043492,4045841,4048347,4057323,4072774,4076999,4078430,4081038,4081165,4090893,4092183,4095910,4097153,4097469,4097922,4097929,4098721,4098728,4098752,4102238,4102827,4103475,4103861,4104258,4104414,4107914,4108824,4109316,4109490,4110656,4111590,4113217,4113493,4115697,4116724,4116731,4116926,4116927,4116935,4117328,4117980,4118226,4118437,4118798,4119032,4119034,4119036,4119272,4119740,4120415,4120660,4120916,4121035,4121081,4121175,4122354,4123641,4124091,4124173,4124411,4124497,4124850,4124893,4125837,4125961,4126205,4127473,4127555,4127744,4127775,4127890,4127899,4128231,4128378,4128586,4128875,4129520,4129546,4129555,4129633,4131073,4131247,4131285,4131834,4132566,4132591,4132897,4132986,4133090,4133113,4133245,4133396,4133416,4133509,4133563,4134270,4135088,4135210,4135249,4135268,4135300,4135576,4135926,4135954,4135968,4136137,4136204,4136222,4136272,4136549,4136604,10200083,16200027,19003868,19003881,19003900,19003901,19003907,19003909,19003910,19003913,19003927',authuser:0,kscs:'c9c918f0_tJjnWbLaJcuV0gKZpJzoDw',u:'c9c918f0',kGL:'US'};google.kHL='en';})();(function(){google.lc=[];google.li=0;google.getEI=function(a){for(var b;a&&(!a.getAttribute||!(b=a.getAttribute("eid")));)a=a.parentNode;return b||google.kEI};google.getLEI=function(a){for(var b=null;a&&(!a.getAttribute||!(b=a.getAttribute("leid")));)a=a.parentNode;return b};google.https=function(){return"https:"==window.location.protocol};google.ml=function(){return null};google.wl=function(a,b){try{google.ml(Error(a),!1,b)}catch(d){}};google.time=function(){return(new Date).getTime()};google.log=function(a,b,d,c,g){if(a=google.logUrl(a,b,d,c,g)){b=new Image;var e=google.lc,f=google.li;e[f]=b;b.onerror=b.onload=b.onabort=function(){delete e[f]};google.vel&&google.vel.lu&&google.vel.lu(a);b.src=a;google.li=f+1}};google.logUrl=function(a,b,d,c,g){var e="",f=google.ls||"";d||-1!=b.search("&ei=")||(e="&ei="+google.getEI(c),-1==b.search("&lei=")&&(c=google.getLEI(c))&&(e+="&lei="+c));c="";!d&&google.cshid&&-1==b.search("&cshid=")&&(c="&cshid="+google.cshid);a=d||"/"+(g||"gen_204")+"?atyp=i&ct="+a+"&cad="+b+e+f+"&zx="+google.time()+c;/^http:/i.test(a)&&google.https()&&(google.ml(Error("a"),!1,{src:a,glmm:1}),a="");return a};}).call(this);(function(){google.y={};google.x=function(a,b){if(a)var c=a.id;else{do c=Math.random();while(google.y[c])}google.y[c]=[a,b];return!1};google.lm=[];google.plm=function(a){google.lm.push.apply(google.lm,a)};google.lq=[];google.load=function(a,b,c){google.lq.push([[a],b,c])};google.loadAll=function(a,b){google.lq.push([a,b])};}).call(this);google.f={};var a=window.location,b=a.href.indexOf("#");if(0<=b){var c=a.href.substring(b+1);/(^|&)q=/.test(c)&&-1==c.indexOf("#")&&a.replace("/search?"+c.replace(/(^|&)fp=[^&]*/g,"")+"&cad=h")};</script><style>#gbar,#guser{font-size:13px;padding-top:1px !important;}#gbar{height:22px}#guser{padding-bottom:7px !important;text-align:right}.gbh,.gbd{border-top:1px solid #c9d7f1;font-size:1px}.gbh{height:0;position:absolute;top:24px;width:100%}@media all{.gb1{height:22px;margin-right:.5em;vertical-align:top}#gbar{float:left}}a.gb1,a.gb4{text-decoration:underline !important}a.gb1,a.gb4{color:#00c !important}.gbi .gb4{color:#dd8e27 !important}.gbf .gb4{color:#900 !important}
</style><style>body,td,a,p,.h{font-family:arial,sans-serif}body{margin:0;overflow-y:scroll}#gog{padding:3px 8px 0}td{line-height:.8em}.gac_m td{line-height:17px}form{margin-bottom:20px}.h{color:#36c}.q{color:#00c}.ts td{padding:0}.ts{border-collapse:collapse}em{font-weight:bold;font-style:normal}.lst{height:25px;width:496px}.gsfi,.lst{font:18px arial,sans-serif}.gsfs{font:17px arial,sans-serif}.ds{display:inline-box;display:inline-block;margin:3px 0 4px;margin-left:4px}input{font-family:inherit}a.gb1,a.gb2,a.gb3,a.gb4{color:#11c !important}body{background:#fff;color:black}a{color:#11c;text-decoration:none}a:hover,a:active{text-decoration:underline}.fl a{color:#36c}a:visited{color:#551a8b}a.gb1,a.gb4{text-decoration:underline}a.gb3:hover{text-decoration:none}#ghead a.gb2:hover{color:#fff !important}.sblc{padding-top:5px}.sblc a{display:block;margin:2px 0;margin-left:13px;font-size:11px}.lsbb{background:#eee;border:solid 1px;border-color:#ccc #999 #999 #ccc;height:30px}.lsbb{display:block}.ftl,#fll a{display:inline-block;margin:0 12px}.lsb{background:url(/images/nav_logo229.png) 0 -261px repeat-x;border:none;color:#000;cursor:pointer;height:30px;margin:0;outline:0;font:15px arial,sans-serif;vertical-align:top}.lsb:active{background:#ccc}.lst:focus{outl [... cut ...] ref=\"/history\"\u003EWeb History\u003C/a\u003E","psrl":"Remove","sbit":"Search by image","srch":"Google Search"},"nds":true,"ovr":{},"pq":"","refpd":true,"rfs":[],"sbpl":24,"sbpr":24,"scd":10,"sce":5,"stok":"1dXXZQXzcaQQeuITAl2kYZU0jF0"},"d":{},"aWiv7g":{},"YFCs/g":{}};google.x(null,function(){});</script></div></body></html>%               ➜  ~
```
maybe, I just say maybe, [this](https://github.com/curl/curl/blob/edafd52be16e4a64f1ae14ea0de722413eedfea2/lib/http_proxy.c#L231) is what curl did.

dibdot, 2018-08-17:
@sequencer feel free to open a PR if you have a better approach.

closed 2018-08-17 by dibdot