openwrt/packages #4961: vsftpd-tls: D-link DIR860l  -> "500 OOPS: munmap"

suste, 2017-10-17:
The vsftpd server is running at first trial , but does not let mi in. The error massage: "500 OOPS: munmap".
Neither from outside nor from inside with "ftp" user.
After the server restart the vsftpd process does not run and can not be started.
Tested with Openwrt-15.05.1, LEDE-17.01.2, LEDE-17.01.3 
No problem with other architecture with same configs (e.g. ar71xx).

/etc/config/vsftpd:
`background=YES
listen=YES
anonymous_enable=NO
local_enable=YES
write_enable=YES
local_umask=022
check_shell=NO
session_support=NO
userlist_enable=YES
userlist_deny=NO
userlist_file=/etc/vsftpd.users
user_config_dir=/etc/vsftpd
chroot_local_user=YES
chroot_list_enable=YES
chroot_list_file=/etc/vsftpd.chroot_list
allow_writeable_chroot=YES
use_localtime=YES`

/etc/vsftpd.chroot_list: empty

/etc/vsftpd.users:
`ftp`

/etc/vsftpd/ftp:
`write_enable=yes`

diizzyy, 2017-12-12:
@suste 
Does it also occur on trunk/master?

suste, 2018-01-01:
Hi!

I do not know.

I tried only stable versions: OWRT-CC-15.05.1, LEDE-17.01.1 / 2/3/4

Regards!
suste

diizzyy, 2019-02-09:
@suste 
Can you give master a try otherwise please close this issue

diizzyy, 2019-03-28:
Closing due to timeout

closed 2019-03-28 by diizzyy