openwrt/packages #4970: <usbip>: hash of usb.ids not updated

tallero, 2017-10-18:
Compile fails due to not updated hash of usb.ids

pprindeville, 2017-10-18:
Can you give more information, like a cut & paste of the failure?

tallero, 2017-10-19:
Solved replacing `HASH` value in `/feeds/packages/net/usbip/Makefile` at line 84 with the following:
`8b52db77343901e5ebe7ae8dabc8a4032d5f0989aa0303fbb77acc1108d154ad`
that should be the hash of http://linux-usb.org/usb.ids

pprindeville, 2017-10-20:
Still would like to see a cut & paste of the failure.  This is standard practice in a bug report.

msbit, 2017-10-22:
@pprindeville take a look at https://github.com/openwrt/packages/pull/4994 for a fix

pprindeville, 2017-10-22:
Still would like to see more about the actual bug.

msbit, 2017-10-22:
@pprindeville fair enough, I'll be doing a few builds of LEDE to check out support for this router, I can put packages back to master to trigger it, and paste output in here.

msbit, 2017-10-22:
```
$ make -j1
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] diffconfig
 make[2] package/cleanup
 make[2] package/compile
...
 make[3] -C package/libs/gettext compile
 make[3] -C package/libs/libiconv compile
 make[3] -C feeds/packages/net/usbip compile
make -r world: build failed. Please re-run make with -j1 V=s to see what's going on
/home/user/development/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 1
```
and:

```
$ make -j1 V=s
make[1]: Entering directory '/home/user/development/source'
make[2]: Entering directory '/home/user/development/source'
make[3]: Entering directory '/home/user/development/source/target/linux'
make[4]: Entering directory '/home/user/development/source/target/linux/ar71xx'
...
make[3]: Entering directory '/home/user/development/source/feeds/packages/net/usbip'
mkdir -p /home/user/development/source/dl
SHELL= flock /home/user/development/source/tmp/.usb.ids.flock -c '      /home/user/development/source/scripts/download.pl "/home/user/development/source/dl" "usb.ids" "47dc941dca801b89f1529e09f0d338755d8fbb0ce75f526e1dcccc4fbf8fc66c" "" "http://www.linux-usb.org/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://www.linux-usb.org/usb.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  577k  100  577k    0     0   228k      0  0:00:02  0:00:02 --:--:--  228k
Hash of the downloaded file does not match (file: 8b52db77343901e5ebe7ae8dabc8a4032d5f0989aa0303fbb77acc1108d154ad, requested: 47dc941dca801b89f1529e09f0d338755d8fbb0ce75f526e1dcccc4fbf8fc66c) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://sources.lede-project.org/usb.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  565k  100  565k    0     0   195k      0  0:00:02  0:00:02 --:--:--  195k
Hash of the downloaded file does not match (file: 0343132c1f0850547ada9b357a7e8babe0c4ec7e936c945f4262fac5412c2f9f, requested: 47dc941dca801b89f1529e09f0d338755d8fbb0ce75f526e1dcccc4fbf8fc66c) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://mirror2.openwrt.org/sources/usb.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://downloads.openwrt.org/sources/usb.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  466k  100  466k    0     0   110k      0  0:00:04  0:00:04 --:--:--  110k
Hash of the downloaded file does not match (file: 312d411c21404a16516cb4d33dcda4cb1c7ed43c003a8736db274de655e418d7, requested: 47dc941dca801b89f1529e09f0d338755d8fbb0ce75f526e1dcccc4fbf8fc66c) - deleting download.
No more mirrors to try - giving up.
Makefile:106: recipe for target '/home/user/development/source/dl/usb.ids' failed
make[3]: *** [/home/user/development/source/dl/usb.ids] Error 2
make[3]: Leaving directory '/home/user/development/source/feeds/packages/net/usbip'
package/Makefile:109: recipe for target 'package/feeds/packages/usbip/compile' failed
make[2]: *** [package/feeds/packages/usbip/compile] Error 2
make[2]: Leaving directory '/home/user/development/source'
package/Makefile:105: recipe for target '/home/user/development/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/user/development/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/development/source'
/home/user/development/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
```

tallero, 2017-10-23:
@pprindeville Sorry for the lack of cut & paste, after correcting it, I couldn't reproduce it because I didn't use git in the first place. It was the one provided by @msbit anyway.

msbit, 2017-10-23:
@tallero it might be worth reopening this until the associated PR is merged

tallero, 2018-09-07:
The issue has been solved in #4994

closed 2018-09-07 by tallero