openwrt/packages #4974: Boost 1.65 does not compile

neheb, 2017-10-19:
Maintainer: @ClaymorePT 
Environment: mvebu, ramips(mt7621)

Description:

The update does not compile. I've simply reverted both commits(update to .1 and to 1.65) but might as well report this.

ClaymorePT, 2017-10-19:
Can you provide more info about the reason why it does not compile? Log output or something?

neheb, 2017-10-19:
looked at the log. seems this error is thrown:

/home/mangix/source/staging_dir/hostpkg/include/python2.7/pyport.h:895:2: error: #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
 #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
  ^
In file included from ./boost/python/detail/is_xxx.hpp:8:0,
                 from ./boost/python/detail/is_auto_ptr.hpp:9,
                 from ./boost/python/detail/copy_ctor_mutates_rhs.hpp:8,
                 from ./boost/python/detail/value_arg.hpp:7,
                 from ./boost/python/object/forward.hpp:10,
                 from ./boost/python/object/pointer_holder.hpp:16,
                 from ./boost/python/to_python_indirect.hpp:10,
                 from ./boost/python/converter/arg_to_python.hpp:10,
                 from ./boost/python/call.hpp:15,
                 from ./boost/python/object_core.hpp:14,
                 from ./boost/python/object.hpp:9,
                 from ./boost/python/list.hpp:10,
                 from libs/python/src/list.cpp:5:

ClaymorePT, 2017-10-19:
It seems that the error comes from Python2.7 source, not from Boost. 
Still, I will investigate what is going on here.

ClaymorePT, 2017-10-19:
@neheb  Can you provide me the specific Target and Subtarget?

I just compiled Boost.Python for target MediaTek Ralink MIPS  and subtarget MT7621. It went out cleanly.
Was this your target and sub-target?

neheb, 2017-10-19:
I tested on mvebu just now. I believe mt7621 was also problematic.

ClaymorePT, 2017-10-19:
@neheb  
For got to add. I tested with LEDE 17.01.4 since OpenWRT master has some issues in compiling the tools.

Can you try using LEDE 17.01.4 ?

neheb, 2017-10-19:
Makes sense. I'm compiling with LEDE master.

ClaymorePT, 2017-10-19:
@neheb  I believe I have found an issue just right now. Looking into it. Sorry for deleting the previous comment...

ClaymorePT, 2017-10-20:
@neheb  False alarm.
At first, I spotted that LEDE 17.04.1 uses a specific revision for the package git repository, where Boost version is at 1.63.0. 
After updating the Makefile for version 1.65.1, I compiled it and it went out clean, so no issues from my side.

Anyway, this might be a problem with the Python 2.7 version, since the error comes from it and not from Boost. 

Can you provide any other information for your build configuration (specific Target and Sub-Target)?

neheb, 2017-10-20:
mvebu Linksys wrt1900ac.

Have only mpd-full which requires boost in my config.

Building on Fedora 26.

ClaymorePT, 2017-10-20:
@neheb 
I just finished compiling LEDE 17.04.1 with Boost 1.65.1 for your target. It went out clean.

`[claymore@manjaro-vm source-17.01.4]$ ls -l ./bin/packages/arm_cortex-a9_vfpv3/packages/
total 8564
-rw-r--r-- 1 claymore users  307239 Oct 20 02:25 alsa-lib_1.1.0-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users    1444 Oct 20 02:41 boost_1.65.1-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   26943 Oct 20 02:41 boost-container_1.65.1-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   70116 Oct 20 02:41 boost-python_1.65.1-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  878566 Oct 20 02:30 glib2_2.50.2-2_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users    6644 Oct 20 02:20 libattr_20160302-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   77736 Oct 20 02:22 libaudiofile_0.3.6-3_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  499237 Oct 20 02:24 libdb47_4.7.25.4.NC-5_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   45701 Oct 20 02:20 libexpat_2.2.0-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  149692 Oct 20 02:21 libfaad2_2.7-3_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users    6471 Oct 20 02:20 libffi_3.2.1-2_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  879133 Oct 20 02:26 libffmpeg-audio-dec_3.2.2-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   72486 Oct 20 02:21 libflac_1.3.1-3_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   16839 Oct 20 02:25 libgdbm_1.11-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   24430 Oct 20 02:20 libid3tag_0.15.1b-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   59967 Oct 20 02:20 libmad_0.15.1b-3_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   24439 Oct 20 02:20 libmms_0.6.4-2_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   28808 Oct 20 02:20 libmpdclient_2.10-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users    9233 Oct 20 02:21 libogg_1.3.2-2_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  155843 Oct 20 02:21 libsndfile_1.0.26-2_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  334041 Oct 20 02:23 libsqlite3_3160000-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   85108 Oct 20 02:21 libupnp_1.6.19-2_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  181982 Oct 20 02:22 libvorbis_1.3.5-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   70632 Oct 20 02:21 libvorbisidec_1.0.3-20150104-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  465374 Oct 20 02:23 libxml2_2.9.4-1_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  192071 Oct 20 02:41 mpd-full_0.19.19-2_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   21488 Oct 20 02:43 Packages
-rw-r--r-- 1 claymore users    6827 Oct 20 02:43 Packages.gz
-rw-r--r-- 1 claymore users   29310 Oct 20 02:43 Packages.manifest
-rw-r--r-- 1 claymore users     151 Oct 20 02:43 Packages.sig
-rw-r--r-- 1 claymore users    1035 Oct 20 02:26 python_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  691272 Oct 20 02:26 python-base_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  788564 Oct 20 02:26 python-codecs_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   38593 Oct 20 02:26 python-compiler_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  100775 Oct 20 02:26 python-ctypes_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  110375 Oct 20 02:26 python-db_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   48232 Oct 20 02:26 python-decimal_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  210315 Oct 20 02:26 python-distutils_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  143349 Oct 20 02:26 python-email_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users    6209 Oct 20 02:26 python-gdbm_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users 1232716 Oct 20 02:26 python-light_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   38183 Oct 20 02:26 python-logging_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   47095 Oct 20 02:26 python-multiprocessing_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   29371 Oct 20 02:26 python-ncurses_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   34150 Oct 20 02:26 python-openssl_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  170500 Oct 20 02:26 python-pydoc_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   43343 Oct 20 02:26 python-sqlite3_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users   54957 Oct 20 02:26 python-unittest_2.7.13-4_arm_cortex-a9_vfpv3.ipk
-rw-r--r-- 1 claymore users  152175 Oct 20 02:26 python-xml_2.7.13-4_arm_cortex-a9_vfpv3.ipk
`

neheb, 2017-10-20:
must be some oddity with fedora then. all i know is reverting the two commits that upgrade it to 1.65 makes it work again.

ClaymorePT, 2017-10-20:
I will try with LEDE master and see where it gets me.

kofec, 2017-10-20:
Hi,
I had similar issue. 
```
CMake Error at [...]/lede/source/staging_dir/host/share/cmake-3.8/Modules/FindBoost.cmake:1842 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path:
  [...]/lede/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/usr/include

  Could not find the following Boost libraries:

          boost_thread
          boost_date_time
          boost_system

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
```
It happen when I just update repository and compile again (all other packages were compile fine). 
I tried 'make clean" didn't help. I tried new repo and "diffconfig" - isssue happen again.

The solution was: Download repository again (fresh) and choose architecture and packages from beginning - not recover from "diffconfig"

ClaymorePT, 2017-10-20:
@neheb   Ok, with LEDE master, I'm now seeing the issue. Checking it out....

ClaymorePT, 2017-10-20:
Ok, this is what I could get.
@neheb  To bypass this issue, you just have to select Python2.7 package and Boost.Python. After that, it will build.
I don't know why this is happening, by I'm going to dig deeper into this.

stintel, 2017-10-20:
FYI, I had issues too before, but I am no longer able to reproduce: #4798

ClaymorePT, 2017-10-20:
@stintel Yes, but those in the Issue you have referenced, came from musl itself. Those issues are out of my control. 
I still rely on people who experience them to tell me, since it is impossible for me to test boost against every single target. I know for a fact that there are some targets to which boost cannot be built.

ClaymorePT, 2017-10-20:
@neheb  Can you please test this commit? -> https://github.com/ClaymorePT/packages/commit/ecb166740c00ca3fd3891a55ad90a91262298721

Thank you!

neheb, 2017-10-20:
> @neheb <https://github.com/neheb> Can you please test this commit? ->
> ClaymorePT/packages@ecb1667
>
> Thank you!

seems to work.

closed 2017-11-04 by thess