openwrt/packages #4975: quassel-irssi: segfault when trying to connect from LEDE 17

TC01, 2017-10-19:
Maintainer: @TC01 (me)
Environment: LEDE 17.02, mvebu (WRT1900ac).

Description: I submitted a package for quassel-irssi some time ago, and confirmed it worked on Chaos Calmer (which I was running at the time). Earlier this year I updated to LEDE and tried to install quassel-irssi from the repository. Unfortunately, when I actually tried to use it to connect to a quassel core (following the instructions [here](https://github.com/phhusson/quassel-irssi)) it segfaulted.

I tried to track down the segfault a few times but failed. I'm filing this bug because I don't really know what else to do, or how else to proceed, in debugging this, so I figured I should open an issue?

There is a pending update (that adds an explicit dependency on openssl, linking the quassel-irssi .so against openssl directly) that I have yet to apply, but the segfault doesn't go away even after doing so.

Note that I also maintain the package in Fedora, and have had no issues, so I'm presuming the failure is somehow related to how the package is built in LEDE. Any advice or help would be greatly appreciated.

diizzyy, 2019-02-09:
@TC01 
Is it still broken on master?

diizzyy, 2019-03-28:
Closing due to timeout

closed 2019-03-28 by diizzyy