openwrt/packages #4983: package "kmod-usb-net" postinst script returned status 255 raspberry pi3

brdelphus, 2017-10-20:
NAME="LEDE"
VERSION="17.01.3, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot 17.01.3"
VERSION_ID="17.01.3"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r3533-d0bf257c46"
LEDE_BOARD="brcm2708/bcm2710"
LEDE_ARCH="arm_cortex-a53_neon-vfpv4"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot 17.01.3 r3533-d0bf257c46"

root@LEDE:/lib/modules/4.4.89# opkg install kmod-usb-net -V 4
opkg_conf_parse_file: Loading conf file /etc/opkg.conf.
opkg_conf_parse_file: Loading conf file /etc/opkg/customfeeds.conf.
opkg_conf_parse_file: Loading conf file /etc/opkg/distfeeds.conf.
pkg_hash_load_feeds: 
pkg_hash_load_feeds: 
pkg_hash_load_feeds: 
pkg_hash_load_feeds: 
pkg_hash_load_status_files: 
pkg_info_preinstall_check: Updating file owner list.
Installing kmod-usb-net (4.4.89-1) to root...
Downloading http://downloads.lede-project.org/releases/17.01.3/targets/brcm2708/bcm2710/packages/kmod-usb-net_4.4.89-1_arm_cortex-a53_neon-vfpv4.ipk
pkg_run_script: Running script /tmp/opkg-kcNcbE/kmod-usb-net-GfFAAe/preinst.
opkg_install_pkg: Installing maintainer scripts.
opkg_install_pkg: Installing data files for kmod-usb-net.
install_data_files: Extracting data files to /.
pkg_write_filelist: Creating //usr/lib/opkg/info/kmod-usb-net.list file for pkg kmod-usb-net.
opkg_install_pkg: Resolving conf files for kmod-usb-net
Unknown package '4'.
opkg_configure_packages: Configuring unpacked packages.
opkg_configure_packages: Reordering packages before configuring them...
Configuring kmod-usb-net.
pkg_run_script: Running script //usr/lib/opkg/info/kmod-usb-net.postinst.
failed to find a module named usbnet
write_status_files_if_changed: Writing status file.
pkg_write_changed_filelists: Saving changed filelists.
Collected errors:
 * opkg_install_cmd: Cannot install package 4.
 * pkg_run_script: package "kmod-usb-net" postinst script returned status 255.
 * opkg_configure: kmod-usb-net.postinst returned 255.

Can't install any usb-ethernet cards, that's pretty pretty bad. Any ideas ?

Whytehorse, 2017-10-27:
Confirmed, however I can use my usb dongle despite the error

KlapusX, 2018-02-08:
no fix ?

brdelphus, 2018-02-08:
No fix what so ever, but i  havent tried newer versions so far. Besides, 
some usb devices are using different modules and they could actually 
work without usb-net, but main problem is, package system gets stuck in 
dependencies problem.

xenek, 2018-04-01:
Issue confirmed on Raspberry Pi 3 B, Very annoying, but doesn't seem to stop things from working.  Bug on LEDE website is low priority, has no one assigned, probably because it's visual only. 

Maybe someone can patch the code to say 'please ignore this error if you have a rasbi!' to the user?  ;-) I spent 20 minutes before I realized things were working even though it failed.

My version: Powered by LuCI lede-17.01 branch (git-17.290.79498-d3f0685) / LEDE Reboot 17.01.4 r3560-79f57e422d

mfp20, 2018-04-17:
Same here: everything working, opkg keeps bugging about kmod-usb-net every time opkg is run. rpi zero, 17.01.4

Is there any way to clean and stop it?

NeoRaider, 2018-04-17:
The issue has been fixed ( https://bugs.openwrt.org/index.php?do=details&task_id=645 , https://bugs.openwrt.org/index.php?do=details&task_id=893 ). Anyways, this is the wrong issue tracker, as kmod-usb-net is a base package and not maintained in https://github.com/openwrt/packages/ .

closed 2018-04-17 by NeoRaider