openwrt/packages #4984: Crelay: segfaulting on ramips

bobafetthotmail, 2017-10-20:
Maintainer: @thess 
Environment: nexx wt3020-8M, ramips/mt7620, LEDE 17.01 and LEDE snapshot.

Description:  crelay segfaults when it tries to access the relay.
So it segfaults immediately when I write **crelay -i**, or after I started it as a daemon with **crelay -d** and I try to access its web interface at <router-ip>:8000.

The same relay works fine in a TL-WR1043N/ND v1 (with crelay), which is an Atheros AR9132.

thess, 2017-10-21:
@ondrej1024 - Are you aware of this?
I'll see if I can reproduce issue in a day or so as soon as I setup a RAMIPS system to test on.

thess, 2017-10-21:
I could use some additional info...

- What type of relay is connected to the system.
- Could you attach the segfault output from your system

ondrej1024, 2017-10-21:
No, this was never reported to me. Seems to be platform related. What be interesting to what line in the code causes the segfault.

bobafetthotmail, 2017-10-21:
> What type of relay is connected to the system.
It's a chinese one, http://www.ebay.co.uk/itm/5V-USB-Relay-2-Channel-Programmable-Computer-Control-For-Smart-Home-/122398788513?hash=item1c7f894ba1:g:gIgAAOSw1WJZGUrN
crelay -i on the kirkwood says
_Detected relay card type is HID API compatible relay card (on 0001:0002:00, 2 channels)_

But as said above, crelay segfaults even without the relay connected, so I doubt the relay is the issue.

I have been running this relay for 4 months (multiple activations daily) on the TL-WR1043N/ND v1, no issues.

>Could you attach the segfault output from your system

I'm using UART here so I see them directly in the serial console, I see the same output also on logread.

a couple attempts at crelay -i  with relay connected:
```
root@LEDE:/# crelay -i
[   45.502321] do_page_fault(): sending SIGSEGV to crelay for invalid read access from 00000018
[   45.519224] epc = 7744ecb4 in libc.so[773d6000+92000]
[   45.529336] ra  = 773bd55d in libusb-1.0.so.0.1.0[773ba000+1a000]
Segmentation fault
root@LEDE:/# crelay -i
[   53.730422] do_page_fault(): sending SIGSEGV to crelay for invalid read access from 00000018
[   53.747307] epc = 7718ccb4 in libc.so[77114000+92000]
[   53.757414] ra  = 770fb55d in libusb-1.0.so.0.1.0[770f8000+1a000]
Segmentation fault
```

and now a couple attempts at crelay -i without any relay connected.
```
root@LEDE:/# crelay -i
[  118.214281] do_page_fault(): sending SIGSEGV to crelay for invalid read access from 00000018
[  118.231184] epc = 773a6cb4 in libc.so[7732e000+92000]
[  118.241294] ra  = 7731555d in libusb-1.0.so.0.1.0[77312000+1a000]
Segmentation fault
root@LEDE:/# crelay -i
[  120.554992] do_page_fault(): sending SIGSEGV to crelay for invalid read access from 00000018
[  120.571895] epc = 770d1cb4 in libc.so[77059000+92000]
[  120.582006] ra  = 7704055d in libusb-1.0.so.0.1.0[7703d000+1a000]
Segmentation fault
```

ondrej1024, 2017-10-21:
Does it segfault also when trying to read the relay. e.g with command: crelay 1

bobafetthotmail, 2017-10-21:
Yes, and presence/absence of relay does not change anything, segfaults in any case.

with the relay connected:

root@LEDE:/# crelay 1
```
[   35.263428] do_page_fault(): sending SIGSEGV to crelay for invalid read access from 00000018
[   35.280317] epc = 77b58cb4 in libc.so[77ae0000+92000]
[   35.290426] ra  = 77ac755d in libusb-1.0.so.0.1.0[77ac4000+1a000]
Segmentation fault
```

without the relay connected

```
root@LEDE:/# crelay 1
[   54.406049] do_page_fault(): sending SIGSEGV to crelay for invalid read access from 00000018
[   54.422951] epc = 77ce7cb4 in libc.so[77c6f000+92000]
[   54.433062] ra  = 77c5655d in libusb-1.0.so.0.1.0[77c53000+1a000]
Segmentation fault
```

bobafetthotmail, 2017-10-21:
The only command I can send without getting segfaults immediately is **crelay** (shows help text) 

**crelay -d** or **crelay -D**, start the daemon fine, but then segfault when I try to access crelay's webinterface from the browser.

thess, 2017-10-21:
With the device plugged into the failing system, could you invoke
> $ lsusb -v 

and post the output?
You might have to install **usbutils**. If you don't have enough room to do this, don't bother.

I think the problem may be in one of the access libraries (hidapi or libusb).

bobafetthotmail, 2017-10-21:
libusb errors out. (yes I installed usbutils)

```
root@LEDE:/# lsusb -v
unable to initialize libusb: -99
```

thess, 2017-10-21:
OK - next step is to figure out what is mis-configured/missing with the USB sub-system.

- Look at your installed packages (``opkg list-installed``) for kmod-usb-xxx and kmod-hid-xxx modules
- ``lsmod`` - looking for usbxxx, hidxxx, evdev, etc.
- Also, look thru (and maybe post here) the output from ``dmesg`` after a fresh boot (looking for USB errors, or more precisely, lack of any usb status info).

The fix may be as simple as installed kernel USB and HID support for your device. My guess is that USB support is missing from kernel.

bobafetthotmail, 2017-10-21:
Well, I have kmod-usb-core and storage, I've selected them when building. I don't see kmod-usb2 and kmod-echi though. 

```
root@LEDE:/# opkg list-installed
adblock - 3.0.3-1
base-files - 176-r5117-fbde9ac
block-mount - 2017-07-31-8ab4fda6-1
busybox - 1.27.2-1
crelay - 0.10.1-1
ddns-scripts - 2.7.6-18
dnsmasq - 2.78-2
dropbear - 2017.75-4
e2fsprogs - 1.43.6-1
fdisk - 2.30.2-1
firewall - 2017-05-27-a4d98aea-1
fstools - 2017-07-31-8ab4fda6-1
fwtool - 1
hidapi - 0.8.0-rc1-2
hostapd-common - 2017-08-24-c2d4f2eb-3
htop - 2.0.2-1
ip-tiny - 4.13.0-1
ip6tables - 1.6.1-1
iptables - 1.6.1-1
iptables-mod-conntrack-extra - 1.6.1-1
iptables-mod-ipopt - 1.6.1-1
iw - 4.9-1
jshn - 2017-09-29-632688e8-1
jsonfilter - 2016-07-02-dea067ad-1
kernel - 4.9.57-1-987915bbf400728edb43fa1850856244
kmod-cfg80211 - 4.9.57+2017-10-06-1
kmod-crypto-crc32c - 4.9.57-1
kmod-crypto-hash - 4.9.57-1
kmod-eeprom-93cx6 - 4.9.57-1
kmod-fs-ext4 - 4.9.57-1
kmod-gpio-button-hotplug - 4.9.57-2
kmod-ifb - 4.9.57-1
kmod-ip6tables - 4.9.57-1
kmod-ipt-conntrack - 4.9.57-1
kmod-ipt-conntrack-extra - 4.9.57-1
kmod-ipt-core - 4.9.57-1
kmod-ipt-ipopt - 4.9.57-1
kmod-ipt-nat - 4.9.57-1
kmod-leds-gpio - 4.9.57-1
kmod-lib-crc-ccitt - 4.9.57-1
kmod-lib-crc-itu-t - 4.9.57-1
kmod-lib-crc16 - 4.9.57-1
kmod-mac80211 - 4.9.57+2017-10-06-1
kmod-mt76 - 4.9.57+2017-10-12-37654d70-1
kmod-mt76-core - 4.9.57+2017-10-12-37654d70-1
kmod-mt7603 - 4.9.57+2017-10-12-37654d70-1
kmod-mt76x2 - 4.9.57+2017-10-12-37654d70-1
kmod-nf-conntrack - 4.9.57-1
kmod-nf-conntrack6 - 4.9.57-1
kmod-nf-ipt - 4.9.57-1
kmod-nf-ipt6 - 4.9.57-1
kmod-nf-nat - 4.9.57-1
kmod-nls-base - 4.9.57-1
kmod-ppp - 4.9.57-1
kmod-pppoe - 4.9.57-1
kmod-pppox - 4.9.57-1
kmod-rt2800-lib - 4.9.57+2017-10-06-1
kmod-rt2800-mmio - 4.9.57+2017-10-06-1
kmod-rt2800-pci - 4.9.57+2017-10-06-1
kmod-rt2800-soc - 4.9.57+2017-10-06-1
kmod-rt2x00-lib - 4.9.57+2017-10-06-1
kmod-rt2x00-mmio - 4.9.57+2017-10-06-1
kmod-rt2x00-pci - 4.9.57+2017-10-06-1
kmod-sched-cake - 4.9.57+2017-01-28-9789742c-1
kmod-sched-core - 4.9.57-1
kmod-scsi-core - 4.9.57-1
kmod-slhc - 4.9.57-1
kmod-udptunnel4 - 4.9.57-1
kmod-udptunnel6 - 4.9.57-1
kmod-usb-core - 4.9.57-1
kmod-usb-storage - 4.9.57-1
kmod-wireguard - 4.9.57+0.0.20171017-1
lede-keyring - 2017-01-20-a50b7529-1
libblkid - 2.30.2-1
libblobmsg-json - 2017-09-29-632688e8-1
libc - 1.1.16-1
libext2fs - 1.43.6-1
libfdisk - 2.30.2-1
libftdi1 - 1.3-1
libgcc - 5.5.0-1
libip4tc - 1.6.1-1
libip6tc - 1.6.1-1
libiwinfo - 2017-08-23-c1a03e82-1
libiwinfo-lua - 2017-08-23-c1a03e82-1
libjson-c - 0.12.1-1
libjson-script - 2017-09-29-632688e8-1
liblua - 5.1.5-1
libmbedtls - 2.6.0-1
libmnl - 1.0.4-1
libncurses - 6.0-1
libnl-tiny - 0.1-5
libpthread - 1.1.16-1
librt - 1.1.16-1
libsmartcols - 2.30.2-1
libubox - 2017-09-29-632688e8-1
libubus - 2017-02-18-34c6e818-1
libubus-lua - 2017-02-18-34c6e818-1
libuci - 2017-09-29-5ad59ad4-1
libuci-lua - 2017-09-29-5ad59ad4-1
libuclient - 2017-09-06-24d6eded-1
libusb-1.0 - 1.0.21-1
libustream-mbedtls - 2016-07-02-ec80adaa-2
libuuid - 2.30.2-1
libxtables - 1.6.1-1
logd - 2017-09-01-b1bc8d5f-1
lua - 5.1.5-1
luci - git-17.293.32122-824b935-1
luci-app-adblock - git-17.293.32122-824b935-1
luci-app-ddns - 2.4.8-2
luci-app-firewall - git-17.293.32122-824b935-1
luci-app-samba - git-17.293.32122-824b935-1
luci-app-sqm - 1.1.3-1
luci-base - git-17.293.32122-824b935-1
luci-lib-ip - git-17.293.32122-824b935-1
luci-lib-jsonc - git-17.293.32122-824b935-1
luci-lib-nixio - git-17.293.32122-824b935-1
luci-mod-admin-full - git-17.293.32122-824b935-1
luci-proto-ipv6 - git-17.293.32122-824b935-1
luci-proto-ppp - git-17.293.32122-824b935-1
luci-proto-relay - git-17.293.32122-824b935-1
luci-proto-wireguard - git-17.293.32122-824b935-1
luci-ssl - git-17.293.32122-824b935-1
luci-theme-bootstrap - git-17.293.32122-824b935-1
mtd - 21
nano - 2.8.7-1
netifd - 2017-10-19-0f96606b-1
odhcp6c - 2017-09-05-51733a6d-4
odhcpd - 2017-09-16-f0bce9cd-1
opkg - 2017-07-28-4bd8601e-1
ppp - 2.4.7-12
ppp-mod-pppoe - 2.4.7-12
procd - 2017-09-28-ee582d16-1
px5g-mbedtls - 4
relayd - 2016-02-07-ad0b25ad-2
resize2fs - 1.43.6-1
rpcd - 2016-12-03-0577cfc1-1
rpcd-mod-rrdns - 20170710
rt2800-pci-firmware - 2017-09-06-a61ac5cf-1
samba36-server - 3.6.25-8
sqm-scripts - 1.1.3-1
swconfig - 11
tc - 4.13.0-1
terminfo - 6.0-1
tune2fs - 1.43.6-1
ubox - 2017-09-01-b1bc8d5f-1
ubus - 2017-02-18-34c6e818-1
ubusd - 2017-02-18-34c6e818-1
uci - 2017-09-29-5ad59ad4-1
uclient-fetch - 2017-09-06-24d6eded-1
uhttpd - 2017-08-19-3fd58e9b-1
uhttpd-mod-ubus - 2017-08-19-3fd58e9b-1
usbutils - 007-6
usign - 2015-07-04-ef641914-1
wireguard-tools - 0.0.20171017-1
wpad-mini - 2017-08-24-c2d4f2eb-3
```

Aaand by installing kmod-usb2 (and its dependencies), crelay is working fine. So I'm closing this issue, as it's unrelated.

Looking at LEDE buildsystem, it seems that the USB support packages aren't installed by default for this device.
https://github.com/lede-project/source/blob/master/target/linux/ramips/image/mt7620.mk#L469
Will test this and then probably send a PR to the LEDE repo.

Thanks for the assistance everyone! :)

closed 2017-10-21 by bobafetthotmail