openwrt/packages #5021: netdata: update version on branch lede-17.01

chrisob, 2017-10-26:
Maintainer: @diizzyy 
Environment: lede-17.01.4-brcm47xx-mips74k running on ASUS RT-N66u

Description:

The netdata version found on the master branch is current (1.8.0), but the version found on the lede-17.01 branch is a bit outdated (1.4.0): https://github.com/openwrt/packages/tree/lede-17.01/admin/netdata

Is there any chance of updating the netdata package for lede as well?

Thanks!
-Chris

diizzyy, 2017-10-26:
@chrisob 
While I don't see an issue doing so I don't use the 17.X branch at all so someone else needs to test and possibly backport it. That said, as far as I know security and severe bug fixes are the only reasons to bump packages in a stable release.

chrisob, 2017-10-27:
@diizzyy 
OK, understood.  

However, I'm not sure if it counts for anything, but I was able to successfully merge the package Makefile/config/patch changes from the master branch into the 17.x branch, build the package, install it, and successfully test/use it.

I've created a PR here: https://github.com/openwrt/packages/pull/5024
Feel free to reject it if you decide it needs more rigorous testing than me confirming that the collectors, webapp, and API is working.

diizzyy, 2017-12-12:
@chrisob 
Can you please close this issue?

closed 2017-12-12 by chrisob