openwrt/packages #5022: mwan3: dead on 4.9.58 kernel

teqlee, 2017-10-26:
I'm install snapshot r5175 on ERL3, mwan3 not work.
My config work well on 17.01.4.

```
Interface status:
 interface wan is error and tracking is active
 interface wan2 is error and tracking is active
```
Check IP rules:
```
Missing 1 of the 2 required interface IP rules
Rules found:
1001:	from all iif pppoe-wan lookup main
```
```
root@ERL3:~# mwan3 restart
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0x100/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0x200/0xff00' to 'fwmark'
```

Troubleshooting:
```
OpenWrt - LEDE Reboot SNAPSHOT r5175-1dcd8e7
LuCI - git-17.298.65971-f24ae58

mwan3 - 2.6.6-1
mwan3-luci - git-17.298.65971-f24ae58-1

Output of "cat /etc/config/mwan3" : 

config policy 'secure_ports'
	list use_member 'wan_m1_w3'
	list use_member 'wan2_m1_w2'
	option last_resort 'unreachable'

config policy 'balanced'
	list use_member 'wan_m1_w3'
	list use_member 'wan2_m1_w3'
	option last_resort 'unreachable'

config interface 'wan'
	option down '3'
	option initial_state 'online'
	option family 'ipv4'
	option track_method 'ping'
	option size '56'
	option flush_conntrack 'never'
	option reliability '1'
	option count '2'
	option timeout '3'
	option interval '60'
	option failure_interval '10'
	option recovery_interval '10'
	option up '3'
	option enabled '1'
	list track_ip 'x.x.x.x'

config interface 'wan2'
	option reliability '1'
	option down '3'
	option initial_state 'online'
	option family 'ipv4'
[... 246 lines cut ...]

Chain mwan3_policy_balanced (1 references)
 pkts bytes target     prot opt in     out     source               destination         
   21  1276 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 statistic mode random probability 0.50000000000 /* wan2 3 6 */ MARK xset 0x200/0xff00
   25  1500 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 /* wan 3 3 */ MARK xset 0x100/0xff00

Chain mwan3_policy_secure_ports (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 statistic mode random probability 0.39999999991 /* wan2 2 5 */ MARK xset 0x200/0xff00
    1    60 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 /* wan 3 3 */ MARK xset 0x100/0xff00

Chain mwan3_rule_https (1 references)
 pkts bytes target     prot opt in     out     source               destination         
  185 11100 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 MARK xset 0x200/0xff00
  185 11100 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x200/0xff00 ! match-set mwan3_sticky_https src,src MARK and 0xffff00ff
  185 11100 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 MARK xset 0x100/0xff00
    1    60 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x100/0xff00 ! match-set mwan3_sticky_https src,src MARK and 0xffff00ff
    1    60 mwan3_policy_secure_ports  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00
  185 11100 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 del-set mwan3_sticky_https src,src
  185 11100 SET        all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match ! 0xfc00/0xfc00 add-set mwan3_sticky_https src,src

Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination         
  185 11100 mwan3_rule_https  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport sports 0:65535 multiport dports 443,22,993,995,465,587 mark match 0x0/0xff00 /* https */
   46  2776 mwan3_policy_balanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 /* default_rule */

teqlee, 2017-11-01:
Work again on snapshot r5217. Tks.

noname77, 2017-11-13:
I am on r5297 and seeing the same behaviour / error:

```
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
```

were you able to determine what was causing this other than just that r5217 fixed this?

Thanks

azuwis, 2017-11-28:
@noname77 I've found the causing of the problem, https://git.lede-project.org/?p=source.git;a=commit;h=1cec4d4ef0145b48094ec0df40c18b8fbb315cf0

    # opkg whatprovides -A ip
    What provides ip
        busybox
        ip-full
        ip-tiny

`ip` applet of busybox does not work with mwan3, installing ip-tiny or ip-full should fix the problem.

noname77, 2017-11-28:
@azuwis thanks for the suggestion, but unfortunately i already have these installed.

```
# opkg whatprovides -A ip
What provides ip
    busybox
    ip-full
    ip-tiny
```
and still seeing the `ip` complaining (output below is repeated for every interface configured in mwan3)

```
# mwan3 restart
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0x100/0xff00' to 'fwmark'
uci: Entry not found
uci: Entry not found
```

I am using davidc502 builds from https://davidc502sis.dynamic-dns.net/ on Linksys WRT1900ACS with LEDE r5422, will try to seek some help there too.

kernel and version info:
```
# cat /proc/version 
Linux version 4.9.65 (davidc502@pc2) (gcc version 5.5.0 (LEDE GCC 5.5.0 r5422-9fe59abef8)
```

Thanks

azuwis, 2017-11-28:
@noname77 Install ip-tiny or ip-full, which means:

    # opkg update
    # opkg install ip-tiny

`opkg whatprovides -A ip` will just list which packages provide the `ip` virtual package.

noname77, 2017-11-28:
@azuwis thanks a lot, that fixed it.

misunderstood what `opkg whatprovides` does

teqlee, 2017-11-28:
@azuwis 
Thanks for pointing exactly. The problem is due to lack of ip-tiny/ip-full package.
I'll reopen this issue.

Ricty, 2017-12-03:
I have the same problem after updating LEDE on my route.
`“interface wan is error and tracking is active” `
And the traffic of wan(s) only depends on the metric regardless of setting in mwan3.
Mwan3 can work again after ip-tiny installed. :)

feckert, 2017-12-07:
Do we have to change the DEPENDS from ip to ip-full | ip-tiny?
On my system i always install ip-full so i have not seen this error before.
https://github.com/openwrt/packages/blob/master/net/mwan3/Makefile#L22

teqlee, 2017-12-10:
@feckert 
Yes, please.

closed 2017-12-10 by teqlee

camelcamro, 2018-01-23:
still have the same issue on current trunk (23.1.2018) 
:(
is it not possible to install all needed packages on trunk version ?

feckert, 2018-01-23:
@camelcamro from my point off view mwan3 will install all needed packages as dependency.
Only ip-full must be installed per hand until following pull request is merged
See **https://github.com/openwrt/packages/pull/5092**

davidmpc, 2021-05-24:
Very helpful.