openwrt/packages #5030: obfsproxy - python issue ?

reinerotto, 2017-10-28:
I built obfsproxy from most recenet LEDE trunk, included it into my custom image.
However, after /etc/init.d/obfsproxy start, it dies immediately.
Getting no error output in logread, I tried a manual start and got
root@LEDE:/# obfsproxy --log-min-severity=info obfs2 --dest=127.0.0.1:1194 server 0.0.0.0:443
Traceback (most recent call last):
  File "/usr/bin/obfsproxy", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/tmp/pip-build-iGrcJj/setuptools/pkg_resources/__init__.py", line 3017, in <module>
  File "/tmp/pip-build-iGrcJj/setuptools/pkg_resources/__init__.py", line 3003, in _call_aside
  File "/tmp/pip-build-iGrcJj/setuptools/pkg_resources/__init__.py", line 3030, in _initialize_master_working_set
  File "/tmp/pip-build-iGrcJj/setuptools/pkg_resources/__init__.py", line 659, in _build_master
  File "/tmp/pip-build-iGrcJj/setuptools/pkg_resources/__init__.py", line 967, in require
  File "/tmp/pip-build-iGrcJj/setuptools/pkg_resources/__init__.py", line 853, in resolve
pkg_resources.DistributionNotFound: The 'obfsproxy==0.2.13' distribution was not found and is required by the application

Any ideas ?

From Makefile:
....
PKG_NAME:=obfsproxy
PKG_VERSION:=0.2.13
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://pypi.python.org/packages/source/o/obfsproxy
PKG_HASH:=1e26c2faef1cfcf856ddf60e9647058a7c78fb0d47f05b58a0f847ed7cc41a66

PKG_LICENSE:=BSD-3-Clause
PKG_LICENSE_FILES:=LICENSE
PKG_MAINTAINER:=Jeffery To <jeffery.to@gmail.com>

include $(INCLUDE_DIR)/package.mk
$(call include_mk, python-package.mk)

define Package/obfsproxy
        SECTION:=net
        CATEGORY:=Network
        TITLE:=A pluggable transport proxy written in Python
        URL:=https://www.torproject.org/projects/obfsproxy.html.en
        DEPENDS:=+python-light +python-crypto +python-pyptlib +python-setuptools +python-yaml +twisted
endef

reinerotto, 2017-10-30:
Also broken on release 17.0.4 of LEDE.

jefferyto, 2018-01-06:
I've installed obfsproxy in LEDE 17.01.4, on both a physical router and in a vm, and both appear to work correctly. As for trunk, there is a more general Python issue I will follow up on.

jefferyto, 2018-03-21:
This should be fixed :smile:

closed 2018-08-18 by dibdot