openwrt/packages #5032: wget --post-data does not transmit data

aparcar, 2017-10-28:
I tried the following command from x86/64 snapshot vm build and Debian 9.

`wget –post-data=”distro=lede” “http://192.168.1.3:5000/image-request” -O-`

The server sides receives no data when using the snapshot, with Debian it works as expected `[”distro”, “lede”]`

dangowrt, 2017-10-28:
I reckon you should use libuclient directly instead of relying in the
wget compatibility wrapper (or even worse: busybox-wget...).

See

https://git.lede-project.org/?p=project/uclient.git;a=summary

diizzyy, 2017-10-28:
...or use curl for instance?

aparcar, 2017-10-28:
@diizzyy curl isn't installed per default or am I mistaken?

diizzyy, 2017-10-28:
No, it's an external package. "wget" is a wrapper, it's not the GNU wget which is also an external package.

aparcar, 2017-10-28:
Thank you two for the clarification, I'll use uclient or curl. 

However, if "wget" has this option mentioned, should't it be fixed or removed from the help menu?

aparcar, 2017-10-29:
Seems like the encoding "chunked" is in compatible to some web servers. So it's not directly a uclient-fetch issue

closed 2017-10-29 by aparcar