openwrt/packages #5058: mwan3: not working on 4.9.58 r5218-f90f94d2c1

renatogmac, 2017-11-04:
using on TP-LINK Archer C7 v4 -  Kernel Version | 4.9.58

Interface status:
 interface speedy is error and tracking is active
 interface virtua is error and tracking is active

/usr/sbin/mwan3 restart
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0x100/0xff00' to 'fwmark'

Troubleshooting Data:
OpenWrt - LEDE Reboot SNAPSHOT r5218-f90f94d2c1
LuCI - git-17.302.29094-87ec343

mwan3 - 2.6.6-1
mwan3-luci - git-17.302.29094-87ec343-1

Output of "cat /etc/config/mwan3" : 

config interface 'speedy'
	option enabled '1'
	option timeout '2'
	option interval '5'
	option reliability '1'
	option count '2'
	option down '3'
	option initial_state 'offline'
	option family 'ipv4'
	list track_ip 'www.google.com'
	option track_method 'httping'
	option size '56'
	option failure_interval '5'
	option recovery_interval '5'
	option up '1'
	option flush_conntrack 'always'

config interface 'virtua'
	option enabled '1'
	list track_ip '208.67.222.222'
	option reliability '1'
	option count '2'
	option timeout '2'
	option interval '5'
	option down '3'
	option initial_state 'offline'
	option family 'ipv4'
	option track_method 'ping'
	option size '56'
	option failure_interval '5'
	option recovery_interval '3'
	option up '1'
	option flush_conntrack 'always'

config member 'speedy1'
	option interface 'speedy'
	option metric '20'
	option weight '50'

config member 'virtua1'
	option interface 'virtua'
	option metric '20'
	option weight '50'

config policy 'speedy_only'
	list use_member 'speedy1'
	option last_resort 'default'

[... 309 lines cut ...]
 pkts bytes target     prot opt in     out     source               destination         
   12  1522 mwan3_rule_MSSQL  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport sports 0:65535 multiport dports 1433 mark match 0x0/0xff00 /* MSSQL */
    1    52 mwan3_policy_speedy_only  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport sports 0:65535 multiport dports 3389 mark match 0x0/0xff00 /* RDP */
    2  1548 mwan3_rule_VOIP  udp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport sports 0:65535 multiport dports 5060:5070 mark match 0x0/0xff00 /* VOIP */
   53  5083 mwan3_policy_virtua_only  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport sports 0:65535 multiport dports 443 mark match 0x0/0xff00 /* https */
    0     0 mwan3_policy_virtua_only  tcp  --  *      *       0.0.0.0/0            189.40.0.0/16        multiport sports 0:65535 multiport dports 0:65535 mark match 0x0/0xff00 /* timdomain */

Chain qos_Default (0 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK restore mask 0xf
    0     0 qos_Default_ct  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 length 0:500 MARK xset 0x22/0xff
    0     0 MARK       icmp --  *      *       0.0.0.0/0            0.0.0.0/0            MARK xset 0x11/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 tcp spts:1024:65535 dpts:1024:65535 MARK xset 0x44/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 udp spts:1024:65535 dpts:1024:65535 MARK xset 0x44/0xff
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0xff

Chain qos_Default_ct (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 5060:5070 /* VOIP SIP */ MARK xset 0x11/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 10000:20000 /* VOIP-RTP */ MARK xset 0x11/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 3389 /* RDP TS */ MARK xset 0x11/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 3389 /* RDP TS */ MARK xset 0x11/0xff
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0xff
Powered by LuCI Master (git-17.302.29094-87ec343) / LEDE Reboot SNAPSHOT r5218-f90f94d2c1

teqlee, 2017-11-08:
Same problem with snapshot before r5217. I have test on four ar71xx device with snapshot r5217 without issue.
Are you sure update all lastest packages (./scripts/feeds update -a) ?

https://github.com/openwrt/packages/issues/5022

lucize, 2017-11-09:
for me is always online, if I manually run the ping test the result of ping is good (ok when connected, timed out when not), if I restart the script with one interface disconnected only then it will be offline r5290-a6e9d146f2 and today feeds update

feckert, 2017-11-12:
@lucize i have found in your output a strange status info.

> Output of "ip route list table 1-250" :
> 
> No data found

I think there should be two tables info shown Nr.1 speedy and Nr.2 "virtua" entires for the wan interfaces?
This are missing! If the firewall marks the packages to use this routing tables but there is no routing information stored then the default table is used (last_resort was set to default).
This is only a guess!

lucize, 2017-11-12:
@feckert you misquoted be, but is ok. this is what I have

```
Last 50 MWAN systemlog entries. Newest entries sorted at the top :

Sat Nov 11 14:24:17 2017 user.info mwan3track[1821]: Detect ifdown event on interface wan2 (eth1.3)
Sat Nov 11 14:24:17 2017 user.info mwan3[4207]: connection tracking not flushed on interface wan2 (eth1.3) ifdown
Sat Nov 11 14:24:16 2017 user.notice mwan3[4207]: Execute ifdown event on interface wan2 (eth1.3)
Sat Nov 11 14:24:16 2017 user.notice mwan3track[1821]: Interface wan2 (eth1.3) is offline
Sat Nov 11 14:23:03 2017 user.info mwan3[2088]: connection tracking not flushed on interface wan (eth0.2) ifup
Sat Nov 11 14:23:02 2017 user.notice mwan3[2088]: Starting tracker on interface wan (eth0.2)
Sat Nov 11 14:23:01 2017 user.notice mwan3[2088]: Execute ifup event on interface wan (eth0.2)
Sat Nov 11 14:23:00 2017 user.info mwan3[1613]: connection tracking not flushed on interface wan2 (eth1.3) ifup
Sat Nov 11 14:22:58 2017 user.notice mwan3[1613]: Starting tracker on interface wan2 (eth1.3)
Sat Nov 11 14:22:56 2017 user.notice mwan3[1613]: Execute ifup event on interface wan2 (eth1.3)
Sat Nov 11 14:22:51 2017 user.notice mwan3[1124]: Using firewall mask 0xff00
```

```
Interface status:
 interface wan is error and tracking is active
 interface wan2 is error and tracking is active

Current ipv4 policies:
balanced:
 wan (100%)

wan2_only:
 unreachable

wan2_wan:
 wan (100%)

wan_only:
 wan (100%)

wan_wan2:
 wan (100%)

Current ipv6 policies:
balanced:
 unreachable

wan2_only:
 unreachable

wan2_wan:
 unreachable

wan_only:
 unreachable

wan_wan2:
 unreachable

Directly connected ipv4 networks:
 192.168.123.255
 127.0.0.0
 127.0.0.0/8
 224.0.0.0/3
 192.168.1.0
 192.168.123.1
 192.168.123.143
 127.0.0.1
 127.255.255.255
 192.168.1.0/24
 192.168.1.1
 192.168.123.0
 192.168.123.0/24
 192.168.123.144
 192.168.1.255

Directly connected ipv6 networks:
 ff00::/8
 fd26:b382:51c9::/64
 fe80::/64

Active ipv4 user rules:
    0     0 S https  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport sports 0:65535 multiport dports 443 
    8   608 - balanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            

Active ipv6 user rules:
    0     0 S https  tcp      *      *       ::/0                 ::/0                 multiport sports 0:65535 multiport dports 443 
    0     0 - balanced  all      *      *       ::/0                 ::/0
```

but the routing table would be the same

```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG    0      0        0 lo
0.0.0.0         192.168.123.1   0.0.0.0         UG    5      0        0 eth0.2
0.0.0.0         192.168.123.1   0.0.0.0         UG    10     0        0 eth1.3
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 br-lan
192.168.123.0   0.0.0.0         255.255.255.0   U     5      0        0 eth0.2
192.168.123.0   0.0.0.0         255.255.255.0   U     10     0        0 eth1.3
192.168.123.1   0.0.0.0         255.255.255.255 UH    5      0        0 eth0.2
192.168.123.1   0.0.0.0         255.255.255.255 UH    10     0        0 eth1.3
```

now I'm showing an example from the same main router but you can see that all the routes are active

renatogmac, 2017-11-12:
it worked fine after: (1) fresh reinstall using the same firmware file as before and certifing all packages were up-to-date; (2) using 2 ethernet ports for 2 WANs, instead of using DHCP and PPPoE wan interfaces at same ethernet port; (3) added one firewall wan group for each wan interface (4) removed qos package

closed 2017-11-12 by renatogmac

camelcamro, 2017-12-11:
hmm, i have no clue, but still not working.
texsted with current trunk.
before i had LEDE trunk version from 2017-08
and now trunk from 10.12.2017 .. gives the same problem:

they are looking like ONLINE, but in detail i see error :(

Interface status:
 interface 4G is error and tracking is active
 interface VPN_ROUT03 is error and tracking is active
 interface VPN_ROUT04 is error and tracking is active
 interface VPN_ROUT05 is error and tracking is active

camelcamro, 2017-12-11:
if i restart it, then it is showing me:

ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0x400/0xff00' to 'fwmark'
bind: Address in use
could not bind to udp port 123
bind: Address in use
could not bind to udp port 123
bind: Address in use
could not bind to udp port 123
bind: Address in use
could not bind to udp port 123
route: SIOCADDRT: File exists
route: SIOCADDRT: File exists

any ideas ?

feckert, 2017-12-11:
Not sure but on an other issue on github ( i do not know the issue Numbers)
the solution was to install the package ip-full.

F-J-S, 2018-01-30:
hi
I have a similar problem.
In my case is failing the script mwan3.sh, it can´t create the rules in the routing policy database for control the route selection. That's why my router just send all the trafic for the wan with lower metric (and the fail over work correctly, but there is nothing of load balancing)

when I run the script:  (try this with ssh console to see errors) 
```
#mwan3 restart
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0x100/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0xfd00/0xff00' to 'fwmark'
ip: invalid argument '0xfe00/0xff00' to 'fwmark'
ip: invalid argument '0x200/0xff00' to 'fwmark'
```

 The script is using the comand "ip rule add...." to try to create the rules in the RPD (routing policy database)

debuging the script:
```
+ ip -4 rule add pref 2253 fwmark 0xfd00/0xff00 blackhole
error-> ip: invalid argument '0xfd00/0xff00' to 'fwmark'
```

(see line 142 381 395 etc in https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh)

 but look like that the ip command in BusyBox v1.27.2 is not handling 
the mask fwmark (/0xff00) and the  "blackhole" action as it would

This happened 10 times during all the script  for diferent rules

see http://man7.org/linux/man-pages/man8/ip-rule.8.html    

I installed the ip-full package and... voila      (thank you very very much @feckert )
no mor error!!! the rules are now there 
```
~# ip rule show
0:      from all lookup local
1001:   from all iif eth0.2 lookup main
1002:   from all iif eth0.3 lookup main
2001:   from all fwmark 0x100/0xff00 lookup 1
2002:   from all fwmark 0x200/0xff00 lookup 2
2253:   from all fwmark 0xfd00/0xff00 blackhole
2254:   from all fwmark 0xfe00/0xff00 unreachable
32766:  from all lookup main
32767:  from all lookup default
```

I'm a novice with Linux( and with the English language too)
I hope this information is useful for someone else

Im runing the LEDE/OpenWrt SNAPSHOT r5977-a9c65c22a1 for a tplink wr-841n V13