openwrt/packages #5060: php7 imap

lucize, 2017-11-04:
@mhei I would like to add php7-mod-imap, I made the uw-imap build depends, the package is packed but when I try to run I get this error and can't figure out the dependencies

```
 php-cli
PHP Warning:  PHP Startup: Unable to load dynamic library '/usr/lib/php/imap.so' - Error relocating /usr/lib/php/imap.so: __stpcpy_chk: symbol not found in Unknown on line 0

```

pprindeville, 2017-11-04:
> I would like to add php7-mod-imap, I made the uw-imap build depends, the package is packed but when I try to run I get this error and can't figure out the dependencies

@lucize You need to get into `feeds/packages/lang/php7/Makefile` and add `imap` into the `PHP7_MODULES` variable.

lucize, 2017-11-04:
I did everything needed, the final lib file complains, when run on target device

pprindeville, 2017-11-04:
> I did everything needed, the final lib file complains, when run on target device

@lucize What does `/usr/lib/php/` contain on the target?  If it doesn't contain `imap.so` then you didn't do _everything_.

pprindeville, 2017-11-04:
Can you point me at your changes?

lucize, 2017-11-04:
it contains the file, as you can see, the imap.so is complaining

```
/usr/lib/php# ls -la
drwxr-xr-x    2 root     root            30 Nov  4 08:50 .
drwxr-xr-x    9 root     root          2763 Nov  4 08:50 ..
-rwxr-xr-x    1 root     root       3285880 Nov  4 08:50 imap.so

```

lucize, 2017-11-04:
I'll print the diff, but don't have access to the build system right now

but I added the libopenssl needed for imap-ssl 
and added a config option with

```
--with-imap=shared,"($STAGING_DIR)/usr" \
--with-imap-ssl="($STAGING_DIR)/usr"
```

pprindeville, 2017-11-04:
And what happens when you do:

`nm -D /usr/lib/php7/imap.so | grep " U "`

pprindeville, 2017-11-04:
Also, next time if you want packaging help, open it as a Pull Request and not as an Issue.

lucize, 2017-11-04:
is realy `nm` there ?
```
nm -D /usr/lib/php7/imap.so | grep " U "
/bin/ash: nm: not found
```

lucize, 2017-11-04:
also I don't have php7 folder in lib

pprindeville, 2017-11-04:
If you're doing development, you should probably build it for your target, as well as `objcopy` and a few other things like `size` and `file`.

lucize, 2017-11-04:
ok, I'll do a PR and test later about the debug, I tested with strace and gdb, but no useful stuff

lucize, 2017-11-04:
```
open("/usr/lib/php/imap.so", O_RDONLY|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=3285880, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\300\276\1\0\0\0\0\0"..., 960) = 960
mmap(NULL, 3211264, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7f522f8f0000
mmap(0x7f522fbf3000, 53248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x103000) = 0x7f522fbf3000
close(3)                                = 0
open("/lib/libcrypto.so.1.0.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libcrypto.so.1.0.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libcrypto.so.1.0.0", O_RDONLY|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=9172048, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\210\5\0\0\0\0\0"..., 960) = 960
mmap(NULL, 3993600, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7f522f521000
mmap(0x7f522f8d0000, 131072, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1af000) = 0x7f522f8d0000
mmap(0x7f522f8ee000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f522f8ee000
close(3)                                = 0
open("/lib/libssl.so.1.0.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libssl.so.1.0.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libssl.so.1.0.0", O_RDONLY|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=2448336, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0M?\1\0\0\0\0\0"..., 960) = 960
mmap(NULL, 2437120, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7f522f2ce000
mmap(0x7f522f519000, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x4b000) = 0x7f522f519000
close(3)                                = 0
munmap(0x7f522f8f0000, 3211264)         = 0
munmap(0x7f522f521000, 3993600)         = 0
munmap(0x7f522f2ce000, 2437120)         = 0
writev(2, [{iov_base="", iov_len=0}, {iov_base="PHP Warning:  PHP Startup: Unabl"..., iov_len=173}], 2PHP Warning:  PHP Startup: Unable to load dynamic library '/usr/lib/php/imap.so' - Error relocating /usr/lib/php/imap.so: __stpcpy_chk: symbol not found in Unknown on line 0) = 173
writev(2, [{iov_base="\n", iov_len=1}, {iov_base=NULL, iov_len=0}], 2
) = 1

```

pprindeville, 2017-11-04:
Where is `c-client.h` in your source tree, by the way?

lucize, 2017-11-04:
it should be in $(STAGING_DIR)/usr/include/ , can't post the full makefile, I don't have power (don't know for how long) in the location where I was building the package

pprindeville, 2017-11-04:
From your project root, what does:

`find . -name c-client.h -print`

yield?

lucize, 2017-11-04:
it should be in the right location, because before adding the stagingdir to the imap config it would complain about this

```
checking for IMAP support... yes, shared
checking for IMAP Kerberos support... no
checking for IMAP SSL support... shared
checking for utf8_mime2text signature... new
checking for U8T_DECOMPOSE...
checking for pam_start in -lpam... no
checking for crypt in -lcrypt... yes
configure: error: Cannot find rfc822.h. Please check your c-client installation.
```

as I told, right now, can't access the build root, I only have the build output
but after I specified the include dir

```
checking for IMAP support... yes, shared
checking for IMAP Kerberos support... no
checking for IMAP SSL support... /home/build/64bit/staging_dir/target-x86_64_musl/usr
checking for utf8_mime2text signature... new
checking for U8T_DECOMPOSE...
checking for pam_start in -lpam... no
checking for crypt in -lcrypt... yes
checking for OpenSSL version... >= 1.0.1
checking for CRYPTO_free in -lcrypto... yes
checking for SSL_CTX_set_ssl_version in -lssl... yes
checking whether rfc822_output_address_list function present... checking whether build with IMAP works... checking for Firebird support... no
```

pprindeville, 2017-11-04:
Well, I just tried, and mine failed:

```
checking for IMAP support... yes, shared
checking for IMAP Kerberos support... no
checking for IMAP SSL support... shared,/home/philipp/bertram/lede/staging_dir/target-x86_64_musl_powercode-bmu
checking for utf8_mime2text signature... new
checking for U8T_DECOMPOSE... 
configure: error: utf8_mime2text() has new signature, but U8T_CANONICAL is missing. This should not happen. Check config.log for additional information.
Makefile:562: recipe for target '/home/philipp/bertram/lede/build_dir/target-x86_64_musl_powercode-bmu/php-7.1.10/.configured_edc3358433e376d6373dc271580dd17f' failed
make[2]: *** [/home/philipp/bertram/lede/build_dir/target-x86_64_musl_powercode-bmu/php-7.1.10/.configured_edc3358433e376d6373dc271580dd17f] Error 1
make[2]: Leaving directory '/home/philipp/bertram/lede/feeds/packages/lang/php7'
package/Makefile:109: recipe for target 'package/feeds/packages/php7/compile' failed
make[1]: *** [package/feeds/packages/php7/compile] Error 2
```

not sure where `c-client.h` is supposed to come from, maybe from [uw-imap](http://www.washington.edu/imap/) but we don't have a package for that.

lucize, 2017-11-04:
I wrote in the first post that I made wu-imap package too, I'll post it also

pprindeville, 2017-11-04:
Ping me on IRC when I can pull from your fork.

mhei, 2017-11-05:
I tried to create the imap package one year or so ago, also created the uw-imap package etc. I gave up, because it was a nightmare. I switched over to use libcurls built-in imap support which was good enough for my project. Would this be possible for you, too?
But back to your problem: maybe only the loading order of the modules is wrong. Could you try to load the imap module last, e.g. by arranging the names of the ini file so that the imap.ini is order last...

lucize, 2017-11-05:
there is only imap plugin (for testing)
I tried a i386 target and saw that cflags and ccflags where not from the build root (previously tried a x64 target on a x64 host and maybe that's why it didn't complained), I adjusted them and, at least for, i386 `php-cli -i` shows 

```
IMAP c-Client Version 2007f
SSL support => enabled
```
I'll try on a clean tree to see what I get

thanks !

pprindeville, 2017-11-05:
Again, let’s move this conversation to a PR.

pprindeville, 2017-11-05:
> I tried to create the imap package one year or so ago, also created the uw-imap package etc. I gave up, because it was a nightmare.

@mhei From what I can tell, it’s no longer maintained and there are several _ad hoc_ forks only, but no official __upstream__ site with numbered releases, etc.

closed 2017-11-05 by pprindeville