openwrt/packages #5062: dovecot: fails to start after upgrade if it uses pigeonhole (Sieve) plugin

val-kulkov, 2017-11-05:
Maintainer: @lucize 
Attention Pigeonhole maintainer: @MikePetullo 
Environment: arm_cortex-a9, bcm53xx/generic, LEDE Reboot SNAPSHOT r5102-c5f97c9

Description:

If the Pigeonhole plugin is used with Dovecot, Dovecot fails to start after an upgrade:
```
root@LEDE:/etc/dovecot# /usr/sbin/dovecot
doveconf: Error: Couldn't load plugin /usr/lib/dovecot/modules/settings/libmanagesieve_login_settings.so: Module is for different ABI version 2.2.ABIv32(2.2.32) (we have 2.2.ABIv33(2.2.33.1))
doveconf: Error: Couldn't load plugin /usr/lib/dovecot/modules/settings/libmanagesieve_settings.so: Module is for different ABI version 2.2.ABIv32(2.2.32) (we have 2.2.ABIv33(2.2.33.1))
doveconf: Error: Couldn't load plugin /usr/lib/dovecot/modules/settings/libpigeonhole_settings.so: Module is for different ABI version 2.2.ABIv32(2.2.32) (we have 2.2.ABIv33(2.2.33.1))
doveconf: Fatal: Error in configuration file /etc/dovecot/dovecot.conf: protocols: Unknown protocol: sieve
```
Recompiling the Pigeonhole package and then removing and re-installing pigeonhole fixes the issue completely.

It looks like pigeonhole should not be a separate package but rather a compile option for dovecot. Should I suggest a PR?

lucize, 2017-11-05:
Yes, I think should be built as a Dovecot option, last time I update both of them, this time I didn't pay attention.
Please do a PR., doveadm was also requested, maybe an option for that too

Regards!

val-kulkov, 2017-11-21:
@lucize @MikePetullo : avoiding the fatal error on dovecot upgrade if the pigeonhole plugin is installed alongside dovecot requires updates to both dovecot Makefile and pigeonhole Makefile. The updates to pigeonhole Makefile are substantial. It seems that I got both Makefiles working correctly now, and I am ready to submit two PRs, but let me first explain what I did, what changes are necessary and why, and receive your feedback.

For every dovecot version upgrade, pigeonhole binaries must be rebuilt and installed before dovecot is upgraded. Otherwise, dovecot fails to start after upgrade with the fatal error as described above.

Making pigeonhole a compile option for dovecot turned out to be a more complex task than I had expected. While rebuilding both packages from the same Makefile would seem to be a straightforward solution at a first glance, in reality it is far from being straightforward. dovecot and pigeonhole use their own configure/make systems and require separate PKG_BUILD_DIR's. Overriding the default build/install code in a LEDE/OpenWrt Makefile to build packages in two separate PKG_BUILD_DIR's quickly got very messy. For this reason, I abandoned the idea to use one Makefile to build both dovecot and pigeonhole and started looking for ways to synchronise pigeonhole rebuilding upon dovecot version upgrade.

Two issues must be solved with the rebuild of pigeonhole binaries: (1) how to force a rebuild of pigeonhole binaries on a dovecot version change, and (2) how to let opkg know that the pigeonhole package has been upgraded so that opkg can pick it up for upgrade.

Solving (1) is easy if one knows the keyword that makes it work. The keyword is "ABI_VERSION". To force a rebuild of packages that depend on package A at run time, package A must define ABI_VERSION in Package/A section of its Makefile. Therefore, once dovecot's Makefile gets "ABI_VERSION:=$(PKG_VERSION)" in "Package/dovecot" section, pigeonhole binaries indeed are rebuilt automatically every time dovecot version changes. This solves the pigeonhole binaries update issue.

Solving (2) is more complicated because opkg has no concept of ABI_VERSION. For opkg to pick up a package upgrade, the package version must be incremented. Therefore, the pigeonhole package version must change every time the dovecot package version changes. I propose that pigeonhole package version includes both dovecot package version and pigeonhole package version, for example: "pigeonhole_2.2.33.2-0.4.20-2". This way, opkg detects the rebuilt pigeonhole binaries and makes them available for upgrading.

Obviously, we want to avoid the need to manually update the pigeonhole version number on every dovecot version upgrade. If a dovecot version upgrade and a pigeonhole version upgrade are in two different PRs, the updated dovecot will be broken until the pigeonhole's PR in openwrt/packages is merged, which may be days or weeks or months away. The pigeonhole version number update must be automatic and it must be concurrent with the dovecot version upgrade.

The following code solves the automatic version update issue for pigeonhole:
```
PKG_VERSION_PLUGIN:=0.4.20
PKG_VERSION_DOVECOT:=$(shell make --no-print-directory -C ../dovecot/ val.PKG_VERSION V=s)
PKG_VERSION:=$(PKG_VERSION_DOVECOT)-$(PKG_VERSION_PLUGIN)
```
With these updates to dovecot and pigeonhole Makefiles, pigeonhole binaries are automatically rebuilt on every dovecot upgrade and become visible to opkg as an upgrade.

There is one other issue IMO, and it may be a controversial one. I propose that "pigeonhole" package is renamed to "dovecot-pigeonhole". There are several reasons for it:

1. pigeonhole is a plugin. There seems to be a convention to name plugins by adding corresponding suffixes to the parent package name, such as lua-\*, luci-\*, php7-mod-\*, perlbase-\*, transmission-\* etc.
2. when named as "dovecot-pigeonhole", the pigeonhole plugin is displayed right next to dovecot in config manager ("make menuconfig") and is therefore highly visible to anyone who explores the available modules for dovecot.
3. when named as "dovecot-pigeonhole", the package version number consisting of both dovecot and pigeonhole version numbers makes more sense: "dovecot-pigeonhole_2.2.33.2-0.4.20-2" looks more comprehensible than "pigeonhole_2.2.33.2-0.4.20-2" (albeit both look rather ugly).

Please let me what you think.

lucize, 2017-11-22:
For me is ok whatever as long as it works, I only use dovecot so I'm not able to speak about pigeonhole, but if the process is automated then is better, I don't think that a name change is such a big deal, the package anyway was recently introduced.

Regards

val-kulkov, 2017-11-23:
ping @MikePetullo

val-kulkov, 2017-11-25:
@lucize @MikePetullo : thank you, I will submit two separate PRs for dovecot and for pigeonhole shortly.

PR #5183 (2017-12-02): pigeonhole: fix runtime dependency on dovecot's ABI
Maintainer: @MikePetullo
Attention dovecot package maintainer: @lucize
Compile tested: bcm53xx, Buffalo WXR-1900DHP, LEDE Reboot SNAPSHOT r5297-bddffc5
Run tested: bcm53xx, Buffalo WXR-1900DHP, LEDE Reboot SNAPSHOT r5297-bddffc5

Description:

Closes #5062. With ABI_VERSION declaration in dovecot's Makefile introduced in PR #5182, pigeonhole plugin will now be rebuilt on every dovecot upgrade. To announce the rebuilt pigeonhole plugin to opkg, introduce a different model of version numbering. The new version numbering now includes both doveco [... cut]
commit 6c6a40ab (2017-12-02): pigeonhole: fix runtime dependency on dovecot's ABI

Closes #5062. With ABI_VERSION declaration in dovecot's Makefile
introduced in PR #5182, pigeonhole plugin will now be rebuilt on every
dovecot upgrade. To announce the rebuilt pigeonhole plugin to opkg,
introduce a different model of version numbering. The new version
numbering now includes both dovecot and pigeonhole plugin version
numbers. Therefore, the rebuilt pigeonhole plugin will be visible to
opkg whenever dovecot or pigeonhole plugin version changes.

Rename "pigeonhole" to "dovecot-pigeonhole". There are several reasons
for renami [... cut]
closed 2017-12-02 by champtar