openwrt/packages #5063: openvswitch: failure on second build

msbit, 2017-11-05:
Maintainer: @commodo
Environment: LEDE (`master`/`e0bd225`)

Description:

Rebuilding LEDE with a configuration including `CONFIG_PACKAGE_kmod-openvswitch=m` fails with the following:

```
make[3]: Entering directory '/home/user/development/source/feeds/packages/net/openvswitch'
rm -f /home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1/.built
touch /home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1:openvswitch-2.8.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99  -I/home/user/development/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/user/development/source/staging_dir/target-mips_24kc_musl/include -I/home/user/development/source/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/usr/include -I/home/user/development/source/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/include/fortify -I/home/user/development/source/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1:openvswitch-2.8.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99  -I/home/user/development/source/staging_dir/target-mips_24kc_musl/usr/include -I/home/user/development/source/staging_dir/target-mips_24kc_musl/include -I/home/user/development/source/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/usr/include -I/home/user/development/source/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/include/fortify -I/home/user/development/source/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/include " LDFLAGS="-L/home/user/development/source/staging_dir/target-mips_24kc_musl/usr/lib -L/home/user/development/source/staging_dir/target-mips_24kc_musl/lib -L/home/user/development/source/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/usr/lib -L/home/user/development/source/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/lib -znow -zrelro " make  -C /home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1:openvswitch-2.8.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ARCH="mips" PYTHONPATH="" PYTHON="" ;
make[4]: Entering directory '/home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1'
PYTHONPATH=./python":" PYTHONDONTWRITEBYTECODE=yes  ./ovsdb/ovsdb-idlc.in annotate ./vswitchd/vswitch.ovsschema ./lib/vswitch-idl.ann > lib/vswitch-idl.ovsidl.tmp && mv lib/vswitch-idl.ovsidl.tmp lib/vswitch-idl.ovsidl
/bin/bash: ./ovsdb/ovsdb-idlc.in: @PYTHON@: bad interpreter: No such file or directory
Makefile:6500: recipe for target 'lib/vswitch-idl.ovsidl' failed
make[4]: *** [lib/vswitch-idl.ovsidl] Error 126
make[4]: Leaving directory '/home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1'
Makefile:271: recipe for target '/home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1/.built' failed
make[3]: *** [/home/user/development/source/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.8.1/.built] Error 2
make[3]: Leaving directory '/home/user/development/source/feeds/packages/net/openvswitch'
package/Makefile:109: recipe for target 'package/feeds/packages/openvswitch/compile' failed
make[2]: *** [package/feeds/packages/openvswitch/compile] Error 2
make[2]: Leaving directory '/home/user/development/source'
package/Makefile:105: recipe for target '/home/user/development/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/user/development/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/development/source'
/home/user/development/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2 
```

The commands to reproduce:

```
git clone https://github.com/lede-project/source.git
pushd source
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig # selecting 'm' for 'Kernel modules' > 'Network Support' > 'kmod-openvswitch'
make
make clean
make -j1 V=s
```

commodo, 2017-11-06:
will take a look 
thanks for reporting

commodo, 2017-11-09:
@msbit
a short term fix for this, would be to change  `python/host` to `python` in the `PKG_BUILD_DEPENDS` var

long-term, i'll try to change some build stuff; the problem with changing  `python/host` to `python` is that you build Python for the host and for the target, even though [for this case and a few other similar cases] only the host Python interpreter is needed

yousong, 2018-06-05:
The issue should be irrelevant now as we use openvswitch.ko from the vanilla kernel build instead of building from the openvswitch source code.

closed 2018-06-05 by yousong