openwrt/packages #5074: net/nfs-kernel-server IPv6 Request

Haringstad, 2017-11-07:
-------------------------------

Maintainer: @tripolar 
Environment: ARM, Used by Entware for QNAP

Description:
Right now, the NFS server doesn't support IPv6. Due to the fact that more and more devices are supporting this, I would kindly ask to enable this? The maintainer of Entware doesn't want to divert at all from your base.

CONFIGURE_ARGS += \
	--disable-gss \
	--disable-nfsv4 \
	--disable-nfsv41 \
	**--disable-ipv6 \** # Please remove this???
	--enable-static \
	--enable-shared \
	--disable-caps \
	--disable-tirpc \
	--disable-nfsdcld

diizzyy, 2017-11-07:
Well we can but since libtirpc, rpcbind isn't in the tree it's a no go.
This should get you going as musl is usually the biggest hurdle... (all distros below uses musl)

https://github.com/openwrt/packages/pull/2858
https://git.archlinux.org/svntogit/packages.git/tree/trunk?h=packages/rpcbind
https://git.alpinelinux.org/cgit/aports/tree/main/rpcbind/
https://git.busybox.net/buildroot/tree/package/rpcbind

https://github.com/openwrt/packages/pull/2857
https://git.archlinux.org/svntogit/packages.git/tree/trunk?h=packages/libtirpc
https://git.alpinelinux.org/cgit/aports/tree/main/libtirpc/
https://github.com/buildroot/buildroot/tree/master/package/libtirpc
https://github.com/gentoo/musl/tree/master/net-libs/libtirpc

tripolar, 2017-11-07:
add the required depencies and create a configurable option for ipv6 and i will push it. feel open the ask for help i think the community (me included ;) ) will be happy to help you :)

Haringstad, 2017-11-14:
Guys, I will do my best?!

PR #6721 (2018-08-12): twisted,netatalk,nfs-kernel-server,snort,lsof,mc,owfs: switch to libt…
Maintainer: me + multiple @jow- @karlp 
Compile tested: arm
Run tested: 

Description:
Switched all packages to libtirpc and removed librpc from packages that build without it. For openwrt base `conntrack-tools`, `busybox `and `rules.mk` needs cleanup as well, should i also open a PR for this or leave it to some core dev? The `portmap` package can than also be removed, since rpcbind should replace it.

If someone could help test the `nfs-kernel-server` + `rpcbind` combo this would be great! I just did a basic test.

fixes:  [... cut]
closed 2018-08-12 by tripolar