openwrt/packages #5075: i2c-tools: Source download fails

jhaakana, 2017-11-07:
Maintainer: @dangowrt 
Environment: ARM Cortex A9 MV88F6720 Rev A0, OpenWRT CC r45001

Description:

The package source is no longer available on fossies.org.

```
PKG_SOURCE_URL:=http://dl.lm-sensors.org/i2c-tools/releases/ \
                http://fossies.org/linux/misc/
```

diizzyy, 2017-11-07:
From what I can tell download URL should be @KERNEL/software/utils/$(PKG_NAME)/ and tarball should be switched to xz.
Preferably bumped to 4.0 while at it (this needs testing)

@jhaakana 
Preferably you should switch to LEDE as the package tree isn't tested against the OpenWrt tree anymore.

jhaakana, 2017-11-07:
@diizzyy Yes, I know that I should move on to LEDE but it is not possible right now (still in few months, I think).

diizzyy, 2018-07-05:
Please close this one, not a supported toolchain
@jhaakana @hnyman @champtar

closed 2018-07-05 by yousong