openwrt/packages #5089: perl: Compile error

DittelHome, 2017-11-08:
I am not able to reopen a issue, please take a look into perl: Compile error #5064 (already closed)

pprindeville, 2017-11-08:
Can you attach logs of how your build is failing and tell us what module is failing?

DittelHome, 2017-11-08:
new update in #5064

pprindeville, 2017-11-08:
Yeah, still not seeing logs.

DittelHome, 2017-11-08:
> Nothing is more useful for a dev than being able to see the errors you're seeing. Please don't raise an issue without being ready to provide this information

Sorry, it was i misunderstanding from my side ! ;)

Here the actuel log:
`make[3]: Entering directory '/home/klaus/lede/feeds/packages/lang/perl-uri'
rm -f /home/klaus/lede/staging_dir/target-mips_24kc_musl/stamp/.perl-uri_installed
(cd /home/klaus/lede/build_dir/target-mips_24kc_musl/perl/URI-1.71; PERL_MM_USE_DEFAULT=1  /home/klaus/lede/staging_dir/hostpkg/usr/bin/perl5.26.0 -MConfig -e '${tied %Config::Config}{cpprun}="mips-openwrt-linux-cpp -E"; unshift(@INC, "."); unless (defined (do "./Makefile.PL")) { if ($@) { die "couldn\047t parse Makefile.PL: $@"; } elsif ($!) { die "Could\047t run Makefile.PL: $!"; } }; die "No Makefile generated!" unless -f "Makefile";'  AR=ar CC=mips-openwrt-linux-gcc CCFLAGS="-D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/klaus/lede/build_dir/target-mips_24kc_musl/perl/URI-1.71:URI-1.71 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/klaus/lede/staging_dir/target-mips_24kc_musl/usr/include -I/home/klaus/lede/staging_dir/target-mips_24kc_musl/include -I/home/klaus/lede/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/usr/include -I/home/klaus/lede/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/include/fortify -I/home/klaus/lede/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/include -D_REENTRANT -D_GNU_SOURCE" CCCDLFLAGS=-fPIC CCDLFLAGS=-Wl,-E DLEXT=so DLSRC=dl_dlopen.xs EXE_EXT=" " FULL_AR=mips-openwrt-linux-ar LD=mips-openwrt-linux-gcc LDDLFLAGS="-shared -rdynamic -L/home/klaus/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/klaus/lede/staging_dir/target-mips_24kc_musl/lib -L/home/klaus/lede/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/usr/lib -L/home/klaus/lede/staging_dir/toolchain-mips_24kc_gcc-5.5.0_musl/lib -znow -zrelro" LDFLAGS="  " LIBC=" " LIB_EXT=.a OBJ_EXT=.o OSNAME=linux OSVERS=2.4.30 RANLIB=: SITELIBEXP=" " SITEARCHEXP=" " SO=so VENDORARCHEXP=" " VENDORLIBEXP=" " SITEPREFIX=/usr INSTALLPRIVLIB=/usr/lib/perl5/5.26 INSTALLSITELIB=/usr/lib/perl5/5.26 INSTALLVENDORLIB=" " INSTALLARCHLIB=/usr/lib/perl5/5.26 INSTALLSITEARCH=/usr/lib/perl5/5.26 INSTALLVENDORARCH=" " INSTALLBIN=/usr/bin INSTALLSITEBIN=/usr/bin INSTALLVENDORBIN=" " INSTALLSCRIPT=/usr/bin INSTALLSITESCRIPT=/usr/bin INSTALLVENDORSCRIPT=" " INSTALLMAN1DIR=/usr/man/man1 INSTALLSITEMAN1DIR=/usr/man/man1 INSTALLVENDORMAN1DIR=" " INSTALLMAN3DIR=/usr/man/man3 INSTALLSITEMAN3DIR=/usr/man/man3 INSTALLVENDORMAN3DIR=" " LINKTYPE=dynamic DESTDIR=/home/klaus/lede/build_dir/target-mips_24kc_musl/perl/URI-1.71/ipkg-install );
Generating a Unix-style Makefile
Writing Makefile for URI
Writing MYMETA.yml and MYMETA.json
Could't run Makefile.PL: Inappropriate ioctl for device at -e line 1.
Makefile:48: recipe for target '/home/klaus/lede/build_dir/target-mips_24kc_musl/perl/URI-1.71/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/klaus/lede/build_dir/target-mips_24kc_musl/perl/URI-1.71/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 25
make[3]: Leaving directory '/home/klaus/lede/feeds/packages/lang/perl-uri'
package/Makefile:109: recipe for target 'package/feeds/packages/perl-uri/compile' failed
make[2]: *** [package/feeds/packages/perl-uri/compile] Error 2
make[2]: Leaving directory '/home/klaus/lede'
package/Makefile:105: recipe for target '/home/klaus/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/klaus/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/klaus/lede'
/home/klaus/lede/include/toplevel.mk:216: die Regel für Ziel „world“ scheiterte
make: *** [world] Fehler 2
`
regards Klaus

pprindeville, 2017-11-08:
editing to use triple-backtick.

pprindeville, 2017-11-08:
Okay, seeing what the issue is... another module whose last statement eval's to undef...

I'm on it.

bazilxp, 2017-11-20:
system("pod2text $WriteMakefileArgs{VERSION_FROM} > README")
    if $developer
        and (not -e 'README' or (stat('README'))[9] < (stat($WriteMakefileArgs{VERSION_FROM}))[9]);
print ("test");         hiding the final error  ....

tmn505, 2017-11-22:
Bumping to 1.72 resolves the issue. Tested on LEDE master.

pprindeville, 2017-11-22:
> Bumping to 1.72 resolves the issue. Tested on LEDE master.

@tmn505 I'm traveling this week.  Can you generate a PR and I'll review and merge it?

Thanks.

closed 2017-11-22 by pprindeville

pprindeville, 2018-02-08:
This should be fixed now with https://github.com/openwrt/packages/pull/5504.  Please confirm.

PR #5165 (2017-11-22): perl-uri: upgrade to 1.72
Fixes #5089

Maintainer: @Naoir @pprindeville 
Compile tested: LEDE/master
commit ccfc8ed0 (2017-11-22): perl-uri: upgrade to 1.72

Fixes #5089