openwrt/packages #5090: bind-server: missing libraries for statistics support

halmartin, 2017-11-08:
Maintainer: @nmeyerhans
Environment: armv6l, pogoplug, LEDE 17.01.1

Description:
The `bind-server` package is being built without support for libxml2 or libjson:
```
CONFIGURE_ARGS += \
	--with-libjson=no \
	--with-libxml2=no
```

I understand this behaviour is likely intentional to keep the size of the package down and reduce the number of dependencies.

However, this choice means that bind statistics do not function, even when libxml2 or libjson is installed on the system.

```
daemon.warn named[]: statistics-channels specified but not effective due to missing XML and/or JSON library
```

There is currently no `bind-server` package available which provides working bind statistics:
```
statistics-channels {
  inet 127.0.0.1 port 8053;
};
```

Accessing `127.0.0.1:8053` with `statistics-channels` enabled in the `named.conf` file results in: **No such URL.** instead of the expected statistics.

`collectd-mod-bind` does not function with the LEDE/OpenWrt `bind-server` package due to `libxml2`/`libjson` being disabled.

Would there be opposition to creating additional variants of the bind-server package to include either libxml2 or libjson so bind statistics work?

nmeyerhans, 2017-11-08:
> Would there be opposition to creating additional variants of the bind-server package to include either libxml2 or libjson so bind statistics work?

Definitely no opposition here. I'm happy to review patches adding support for additional build-time options in the Config interface. Or I'll work on it when I have a little time...

halmartin, 2017-11-08:
> Definitely no opposition here. I'm happy to review patches adding support for additional build-time options in the Config interface.

Awesome, I'll see if I can achieve something sensible and submit a pull request.

halmartin, 2017-11-21:
Pull request here: https://github.com/openwrt/packages/pull/5157

halmartin, 2017-11-25:
Merge request approved, thanks @nmeyerhans !

closed 2017-11-25 by halmartin