openwrt/packages #5097: <ddns-scripts> ddns-scripts_cloudflare.com-v4 not updating subdomain A record, and changes traffic status on success

sinayion, 2017-11-10:
DNS provider: Cloudflare
Device: Linksys WRT1200AC
Firmware: LEDE
Version: LEDE Reboot 17.01.4 r3560-79f57e422d / LuCI lede-17.01 branch (git-17.290.79498-d3f0685)
Packages installed:

- ddns-scripts
- ddns-scripts_cloudflare.com-v4
- curl
- wget

Issue (some names are custom, for the sake of the issue description):
I own mydomain.com, and have been able to successfully change the "mydomain.com" A record via LEDE. If I put a false value on Cloudflare's admin dashboard, and then perform the update via the router, I see it successfully changed within a minute. However, it changes the traffic status from going through Cloudflare (orange cloud icon), to DNS only (grey cloud icon).

Now, I want to update home.mydomain.com via the router, because I am adding more routers at physical locations. I put the settings in LEDE, and specified the Lookup Hostname as "home.mydomain.com" and Domain as "mydomain.com". This time, the A record is not changing.

**tldr:
Issue 1: When Cloudflare's A record is updated, it changes the traffic status from Cloudflare, to DNS only.
Issue 2: I cannot update subdomain A records, only the main domain A record.**

I've attached a log from the router, as well as an image showing the Cloudflare icons mentioned above.

[lede cloudflare logs for home_mydomain_com.txt](https://github.com/openwrt/packages/files/1460496/lede.cloudflare.logs.for.home_mydomain_com.txt)
[image]

chris5560, 2017-11-10:
Hi Harry,
some versions back the syntax of "domain" was changed to get rid of the need to always update/rebuild tld_names.dat.gz

# option domain   - "hostname@yourdomain.TLD"	# syntax changed to remove split_FQDN() function and rebuild tld_names.dat.gz

please try

sinayion, 2017-11-11:
Cheers, @chris5560, that worked! Issue 2 is fixed.

Issue 1 is still there, though, where the change in the A record causes the Cloudflare proxy to turn off. I think I found what is needed to be changed here: https://api.cloudflare.com/#dns-records-for-a-zone-update-dns-record. If I'm reading that right, we need the following variable added to the script:
`"proxied": true`

chris5560, 2017-11-11:
It's easy to add this variable to the script (command send to Cloudflair) but do we need a new ddns-scripts config variable for it ?
Do you know what happens, if this parameter is always set to "true" ?
I'm not familiar with Cloudflair; thanks for support

sinayion, 2017-11-11:
We definitely need this variable added, and also shown in the relevant luci config page, where users can choose between the two values "true" and "false". This is a relevant cloudflare page: https://support.cloudflare.com/hc/en-us/articles/200169626-What-subdomains-are-appropriate-for-orange-gray-clouds-

From that page:
 - **If proxied is set to true:** An orange cloud means your web traffic is running through Cloudflare making your site faster, safer and smarter.
 - **If proxied is set to false:** A gray cloud means your web traffic is not running through Cloudflare.

I would assume that "true" should be the default for users.

dkadioglu, 2018-04-04:
Hi,

I had the same problem and thanks to this issue I successfully added the optional paramter 'false' and now the state of 'proxied' is not changed anymore. But there is another similar problem: When I change the TTL from 'Automatic' to something else on the cloudflare webpage, after the script has successfully updated the IP, the TTL is changed back to 'Automatic'. It would be nice, if this parameter can be kept as it is, too. Could this be fixed on the same way? Or, wouldn't it be better to allow e.g. a comma separated list of such parameters. Maybe something like this in the optional parameter field: proxied=false, ttl=2m (or 2h or 1d or any of the other possible values for ttl).

Thanks in advance and best

pycvalade, 2018-05-08:
@dkadioglu would you mind helping me with the config you used? I've been trying to get this working without luck for a few hours now! 

Here's a dump of the logs:
```
 173939  note : PID '23223' started at 2018-05-08 17:39
 173939       : ddns version  : 2.7.7-5
 173939       : uci configuration:\nddns.cloudflare_ipv4.cacert='/etc/ssl/certs/ca-certificates.crt'
ddns.cloudflare_ipv4.domain='sip@domain.com'
ddns.cloudflare_ipv4.enabled='1'
ddns.cloudflare_ipv4.interface='wan'
ddns.cloudflare_ipv4.ip_network='wan'
ddns.cloudflare_ipv4.ip_source='network'
ddns.cloudflare_ipv4.lookup_host='sip.domain.com'
ddns.cloudflare_ipv4.param_opt='true'
ddns.cloudflare_ipv4.password='*password*'
ddns.cloudflare_ipv4.service_name='cloudflare.com-v4'
ddns.cloudflare_ipv4.use_https='1'
ddns.cloudflare_ipv4.username='user@gmail.com'
ddns.cloudflare_ipv4=service
 173939       : verbose mode  : 0 - run normal, NO console output
 173940       : check interval: 600 seconds
 173940       : force interval: 259200 seconds
 173940       : retry interval: 60 seconds
 173940       : retry counter : 0 times
 173940       : No old process
 173940       : last update: never
 173940       : Detect registered/public IP
 173940       : #> /usr/bin/host -t A sip.domain.com  >/var/run/ddns/cloudflare_ipv4.dat 2>/var/run/ddns/cloudflare_ipv4.err
 173940 ERROR : BIND host error: '1'
 173940       : 
 173940       : Detect registered/public IP
 173941       : #> /usr/bin/host -t A sip.domain.com  >/var/run/ddns/cloudflare_ipv4.dat 2>/var/run/ddns/cloudflare_ipv4.err
 173941 ERROR : BIND host error: '1'
 173941       : 
 173941  WARN : Get registered/public IP for 'sip.domain.com' failed - retry 1/0 in 60 seconds
```

chris5560, 2018-05-13:
If looks like there are problems to lookup the IP for "sip.domain.com".
Does this host exists in World Wide DNS?
Could you "nslookup" or "ping" this host?
Could you "nslookup" or "ping" i.e. google.com?

dkadioglu, 2018-05-15:
@pycvalade puuh, never had that problem. To me the error sounds more like something is wrong with the file containing the ca-certs. Does curl work for any other url, e.g. https://www.google.com or something else via https? Maybe you can try to reinstall the package ca-certificates? Just some ideas to check.

pycvalade, 2018-05-15:
Hi guys!

Thanks for the help, tried a few things, still nothing:
- restarted router, just in case
- removed 0 byte crt: `root@py-net:~# rm /etc/ssl/certs/ca-certificates.crt`
- installed ca-certificates, `/etc/ssl/certs/ca-certificates.crt` doesn't exist after install, but there's a bunch of files in `/etc/ssl/certs/` though...
- tried curl on https, fails because `/etc/ssl/certs/ca-certificates.crt` doesn't exist:
```
root@py-net:~# curl https://www.google.com
curl: (77) Error reading ca cert file /etc/ssl/certs/ca-certificates.crt - mbedTLS: (-0x3E00) PK - Read/write of file failed
```
- curl from router downloads 0 bytes:
`curl -o /etc/ssl/certs/ca-certificates.crt http://curl.haxx.se/ca/cacert.pem`
- download from my computer (on the same network) works..

Anyway.. found a way around all of this:
`scp Downloads/cacert.pem root@10.0.0.1:/etc/ssl/certs/ca-certificates.crt`

Checking back logs, I see success :)
```
 103922       : IPv4 at CloudFlare.com already up to date
 103922  info : Update successful - IP '192.222.194.13' send
 103922  info : Forced update successful - IP: '192.222.194.13' send
 103922       : Waiting 600 seconds (Check Interval)
```

Thanks for your help! Someone might want to look into the ca-certificates bundle? Cheers!

Mosney, 2018-07-05:
> Hi Harry,
some versions back the syntax of "domain" was changed to get rid of the need to always update/rebuild tld_names.dat.gz
> \# option domain   - "hostname@yourdomain.TLD"	# syntax changed to remove split_FQDN() function and rebuild tld_names.dat.gz
>please try  

@sinayion I also have question both your issuse1&2. But I still don't understand how to fix it after read it, so could you explain it specific for me? both of your issue. thanks a lot!

dibdot, 2018-08-08:
fixed in latest snapshot and 18.06 tree with referenced update commit.

closed 2018-08-08 by dibdot