openwrt/packages #5110: gawk 4.2.0:  Install fails, igawk missing

CGretski, 2017-11-12:
```
cp: cannot stat '/usr/src/openwrt/LEDE/build_dir/target-mips_24kc_musl/gawk-4.2.0/ipkg-install/usr/bin/{gawk}': No such file or directory
make[3]: *** [Makefile:47: /usr/src/openwrt/LEDE/build_dir/target-mips_24kc_musl/gawk-4.2.0/.pkgdir/gawk.installed] Error 1
make[3]: Leaving directory '/usr/src/openwrt/LEDE/feeds/packages/utils/gawk'

```
As per gawk NEWS file

>  Changes  from 4.1.4 to 4.2.0
> ---------------------------
> <snip>
> 4. The igawk script and igawk.1 man page are no longer installed by
>    `make install'.  They have been obsolete since gawk 4.0.0.
> 

Reference to igawk needs removing from Makefile
```
define Package/gawk/install
	$(INSTALL_DIR) $(1)/usr/bin
$(CP) $(PKG_INSTALL_DIR)/usr/bin/{gawk,igawk} $(1)/usr/bin/
```

diizzyy, 2017-12-12:
Thanks for reporting

dibdot, 2018-08-18:
seems to be fixed.

closed 2018-08-18 by dibdot