openwrt/packages #5117: Is it possible using second wan connection for recovery and turn back first connection when it is ready?

alptolga, 2017-11-13:
Hi,
I use  mwan3 for recovery. when my first connection fails, my second connection works. But after my first connection becomes ready,  I must turn off my  recovery connection. Is it possible to turn back my primary connection automatically?

feckert, 2017-11-16:
@alptolga This should work out of the box. If it is configured correct. What version do you have.

alptolga, 2017-11-16:
@feckert Thanks for your reply.
I am using  ''mwan3  1.6-2   luci-app-mwan3  1.4-3'' which are in the OpenWrt Chaos Calmer 15.05.1 default package list.

feckert, 2017-11-16:
@alptolga It should work also for 1.6.2 but there are some issue which are fixed on lede-17.01 could you try lede-17.01 branch?

feckert, 2017-12-07:
@alptolga any progress on your setup?

alptolga, 2017-12-09:
I still use OpenWrt Chaos Calmer 15.05.1  with mwan3 as I wrote before. I couldn't find any solution.

feckert, 2017-12-12:
@alptolga sorry this is only supported in the master feed with the option "initial_state"

alptolga, 2017-12-12:
Thanks for your answers and interest about the issue I live. So the only way is closing.

closed 2017-12-12 by alptolga