openwrt/packages #5120: openzwave: missing symbols in shared library

artynet, 2017-11-13:
Maintainer: @stintel @ffainelli @sdwalker 
Environment: LEDE 17.0.1.4 x86, LEDE 17.0.1.2 bcm2710

as reported in the subject, all zwave symbols are missing in the shared library while are normally archived in static one :

```
root@lede-17014 /root [#]# objdump -t /usr/lib/libopenzwave.so.1.4 

/usr/lib/libopenzwave.so.1.4:     file format elf32-i386

SYMBOL TABLE:
no symbols
```

any hints on this ? I have also built the library with full support to UDEV and HID but nothing changes....any help would be really appreciated ! 

Thanks in advance, Arturo

stintel, 2017-11-29:
Seems to be caused by `CONFIG_USE_SSTRIP=y`. Maybe try with `CONFIG_USE_STRIP=y` instead, or build the package manually without stripping at all:
`make package/openzwave/{clean,compile} STRIP=/bin/true`

mhei, 2019-03-26:
I'm closing this issue due to house-keeping and long inactivity. I also think it's not valid because I verified that the output of `objdump -t` against a known-to-work shared library file results in the same output, i.e. it reports no symbols. So I guess that the libopenzwave.so is working, though I haven't checked that since I don't own any hardware for this.

closed 2019-03-26 by mhei