openwrt/packages #5133: lighttpd: mod_mysql_vhost.so not building

artynet, 2017-11-16:
Maintainer: @MikePetullo
Environment: bcm2710, LEDE stable updated to latest commit

Description:

the **mod_mysql_vhost.so** is not built by the enviroment and for this reason the packaging fails. Please take a look at the final part of the log : 

```
cortex-a53_neon-vfpv4/lighttpd-mod-mysql_vhost/usr/lib/lighttpd
cp -fpR /home/artynet/lede-arty/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/lighttpd-1.4.45/ipkg-install/usr/lib/lighttpd/mod_mysql_vhost.so /home/artynet/lede-arty/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/lighttpd-1.4.45/ipkg-arm_cortex-a53_neon-vfpv4/lighttpd-mod-mysql_vhost/usr/lib/lighttpd
cp: cannot stat '/home/artynet/ede-arty/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/lighttpd-1.4.45/ipkg-install/usr/lib/lighttpd/**mod_mysql_vhost.so**': No such file or directory
make[2]: *** [/home/artynet/lede-arty/bin/targets/brcm2708/bcm2710/packages/lighttpd-mod-mysql_vhost_1.4.45-3_arm_cortex-a53_neon-vfpv4.ipk] Error 1
Makefile:174: recipe for target '/home/artynet/lede-arty/bin/targets/brcm2708/bcm2710/packages/lighttpd-mod-mysql_vhost_1.4.45-3_arm_cortex-a53_neon-vfpv4.ipk' failed
make[2]: Leaving directory '/home/artynet/lede-arty/feeds/packages/net/lighttpd'
make[1]: *** [package/feeds/packages/lighttpd/compile] Error 2
package/Makefile:105: recipe for target 'package/feeds/packages/lighttpd/compile' failed
make[1]: Leaving directory '/home/artynet/lede-arty'
make: *** [package/lighttpd/compile] Error 2
```
any hints about that ? thank you very much in advance...

Arturo

lucize, 2017-11-25:
try adding this https://github.com/openwrt/packages/pull/4221 and modify the link in lighttpd against libmariadb instead of mysqlclient

p3x-robot, 2017-12-06:
yeah, that is true, MySQL 5.1 is so old MariaDB 5.5.58 is awesome! The best last of together MySql and MariaDB, after that they change a you but all programs work with MariaDB 5.5.58... 👍