openwrt/packages #5149: macchanger on Banana Pi M1 generates the same addresses

odmdas, 2017-11-18:
Maintainer: @paulfertser, @psycho-nico 
Environment: sunxi Banana Pi M1 LEDE Reboot 17.01.4 r3560-79f57e422d

Description:
macchanger works incorrectly on Banana Pi M1.
It generates always the same MAC with each generation option (-e, -a, -A, -r).
```
root@LEDE:/# /etc/init.d/network stop
root@LEDE:/# macchanger -r eth0
[ERROR] Could not read permanent MAC: Resource busy
Current MAC:   02:93:30:4b:56:ef (unknown)
Permanent MAC: 00:00:00:00:00:00 (XEROX CORPORATION)
New MAC:       5e:f7:e6:77:07:3c (unknown)
root@LEDE:/# macchanger -r eth0
[ERROR] Could not read permanent MAC: Resource busy
Current MAC:   5e:f7:e6:77:07:3c (unknown)
Permanent MAC: 00:00:00:00:00:00 (XEROX CORPORATION)
New MAC:       5e:f7:e6:77:07:3c (unknown)
It's the same MAC!!
```
When this BananaPi runs armbian, it's macchanger works as expected and generates a different MAC on every execution.
Also, I could't reproduce this issue in x86 LEDE build running in virtual machine (KVM).

As a workaround, MAC can be changed in Banana Pi LEDE by init script like in post [https://forum.lede-project.org/t/change-mac-address-problem/4085/11](url)

paulfertser, 2017-11-19:
Hello. Can you please show strace output for those two macchanger invocations?

odmdas, 2017-11-19:
Hello. Here it is:
```
root@LEDE:~# /etc/init.d/network stop
root@LEDE:~# strace -o 1.txt macchanger -r eth0
[ERROR] Could not read permanent MAC: Resource busy
Current MAC:   02:93:3d:01:5e:bd (unknown)
Permanent MAC: 00:00:00:00:00:00 (XEROX CORPORATION)
New MAC:       5e:f7:e6:77:07:3c (unknown)
root@LEDE:~# strace -o 2.txt macchanger -r eth0
[ERROR] Could not read permanent MAC: Resource busy
Current MAC:   5e:f7:e6:77:07:3c (unknown)
Permanent MAC: 00:00:00:00:00:00 (XEROX CORPORATION)
New MAC:       5e:f7:e6:77:07:3c (unknown)
It's the same MAC!!
```
[1.txt](https://github.com/openwrt/packages/files/1485822/1.txt)
[2.txt](https://github.com/openwrt/packages/files/1485823/2.txt)

paulfertser, 2017-11-19:
Very useful,  thank you!
The issue is visible in this:
```
open("/dev/hwrng", O_RDONLY|O_LARGEFILE) = 3
read(3, 0xbee66d54, 4)                  = -1 ENODEV (No such device)
```
and https://github.com/alobbs/macchanger/blob/master/src/main.c#L97

In other words, for some reason you have /dev/hwrng file but the kernel says there's no driver associated with it. Why do you have that file then?

paulfertser, 2017-11-19:
If you remove that file then macchanger won't get confused and should provide correct results. But the question is why it is created in the first place.

paulfertser, 2017-11-19:
Ok, and the answer is that your kernel config has HW_RANDOM enabled for HW_RANDOM_TIMERIOMEM but timeriomem_rng resource is not defined in the device tree file. Most probably.

odmdas, 2017-11-19:
Flash with older system LEDE Reboot 17.01.2 r3435-65eec8bd5f, same problem.
```
root@LEDE:/# rm /dev/hwrng 
root@LEDE:/# /etc/init.d/network stop
root@LEDE:/# strace -o /tmp/strace.txt macchanger -r eth0
```
the process does not end and I have to interrupt it.
[strace.txt](https://github.com/openwrt/packages/files/1485881/strace.txt)

paulfertser, 2017-11-19:
It's just waiting for more entropy, that's normal. Some network traffic will give it, or you can install rng-tools to have a special daemon supply some probably.

closed 2018-08-18 by dibdot