openwrt/packages #5178: ddns-scripts: Add DNSEver to Dynamic DNS

gschaub, 2017-11-26:
Maintainer: @chris5560
Environment: (17.1.04)

Description:

I apologize if this isn't the proper way to request an addition to the ddns-scripts package, but I was hoping you'd perform a update at your convenience to that package to add an additional provider.
```
"dbsever.com"           "http://[USERNAME]:[PASSWORD]@dyna.dnsever.com/update.php?host[[DOMAIN]]"
```
Note that I found the double brackets on domain are correct in that they keep that bracket character for the domain.  The required format is ?host[mydomain.com] rather than an = sign.

chris5560, 2017-11-27:
Only IPv4 or also IPv6

gschaub, 2017-11-27:
This is only IPv4.  I haven’t tested it for v6.  Is that okay to include it for v4 only?

 

Thanks…Greg

chris5560, 2017-11-28:
It's ok for IPv4 only, but if there is IPv6 support we should implement it.

gschaub, 2017-11-28:
I don’t see ddns for AAAA records on their site, but have sent an e-mail to verify.

gschaub, 2017-12-11:
Note that I have verified that dnsever doesn’t currently support ddns for ip6. Please let me know if you plan to merge this into the project services file. 

Thanks...Greg

closed 2018-01-13 by hnyman