openwrt/packages #5181: ffmpeg:ERROR: shine not found using pkg-config

xuing, 2017-11-27:
I'm build LEDE on Ubuntu 17.10.
There are Error Info：
`touch /home/xpf/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-3.2.8/.prepared_f9165894b22b3b77bad3e7e50ca63601_6664517399ebbbc92a37c5bb081b5c53_check
. /home/xpf/lede/include/shell.sh; xzcat /home/xpf/lede/dl/ffmpeg-3.2.8.tar.xz | tar -C /home/xpf/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-3.2.8/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/* /home/xpf/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-3.2.8
touch /home/xpf/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-3.2.8/.prepared_f9165894b22b3b77bad3e7e50ca63601_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/xpf/lede/staging_dir/target-mipsel_24kc_musl/stamp/.ffmpeg.full_installed
( cd /home/xpf/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-3.2.8;
 CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/xpf/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-3.2.8:ffmpeg-3.2.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/xpf/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/xpf/lede/staging_dir/target-mipsel_24kc_musl/include -I/home/xpf/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.5.0_musl/usr/include -I/home/xpf/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.5.0_musl/include/fortify -I/home/xpf/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.5.0_musl/include -fpic" LDFLAGS="-L/home/xpf/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -L/home/xpf/lede/staging_dir/target-mipsel_24kc_musl/lib -L/home/xpf/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.5.0_musl/usr/lib -L/home/xpf/lede/staging_dir/toolchain-mipsel_24kc_gcc-5.5.0_musl/lib -znow -zrelro" ./configure --enable-cross-compile --cross-prefix="mipsel-openwrt-linux-musl-" --arch="mipsel" --target-os=linux --prefix="/usr" --pkg-config="pkg-config" --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-inline-asm --disable-mipsdsp --disable-mipsdspr2 --disable-mipsfpu --disable-msa --disable-mmi --disable-fast-unaligned --disable-runtime-cpudetect --disable-yasm --enable-avresample --enable-libopus --enable-small --enable-libshine )
**ERROR: shine not found using pkg-config**

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
Include the log file "config.log" produced by configure as this will help
solve the problem.
Makefile:686: recipe for target '/home/xpf/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-3.2.8/.configured_f78b6842b4d5492e928f33d107540819' failed
make[3]: *** [/home/xpf/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg-3.2.8/.configured_f78b6842b4d5492e928f33d107540819] Error 1
make[3]: Leaving directory '/home/xpf/lede/feeds/packages/multimedia/ffmpeg'
package/Makefile:109: recipe for target 'package/feeds/packages/ffmpeg/compile' failed
make[2]: *** [package/feeds/packages/ffmpeg/compile] Error 2
make[2]: Leaving directory '/home/xpf/lede'
package/Makefile:105: recipe for target '/home/xpf/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/xpf/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xpf/lede'
/home/xpf/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2`

I try to reinstall pkg-config, but It doesn't work.
How can I fix this? Thanks.

xuing, 2017-11-29:
I disable `libshine `when building `ffmpeg`.
But...I met other error again 

> Package php7-mod-imap is missing dependencies for the following libraries:
> libpam.so.0
> Makefile:602: recipe for target '/home/xpf/lede/bin/packages/mipsel_24kc/packages/php7-mod-imap_7.1.12-1_mipsel_24kc.ipk' failed

xuing, 2017-11-29:
I disable `php7-mod-imap` again.
The issue should probably be closed.
Escaping the problem is the best solution.probably~
anyway,Thanks.

thess, 2017-11-30:
@xuing - ffmpeg build dependencies should be fixed by 7fd7ea45fa03ae30566afa42853383a25975269d

Your __php7-mod-imap__ issue should be submitted as a separate posting.

closed 2017-11-30 by thess