openwrt/packages #5229: libgstreamer1 : Missing librt dependency

thevamad, 2017-12-05:
Maintainer: @MikePetullo
Environment: ar71xx

Description: libgstreamer1 complains about missing librt dependency. Did the following to get it to build

`define Package/libgstreamer1
  $(call Package/gstreamer1/Default)
  TITLE+= library (core)
  DEPENDS+= +glib2 +libpthread +libxml2 **+librt**
  HIDDEN:=1
endef`

thess, 2017-12-06:
Sorry about that - was about to post the following...

We need to know more about your build environment since neither the buildbots nor myself have reproduced this issue. There is usually no direct dependency on librt however, configure seems to find what is necessary even when it is not present in the build.

diizzyy, 2018-07-05:
@thess 
Close, reporter timeout?

closed 2018-07-05 by thess