openwrt/packages #5258: python: if selected building openwrt crashes

nerdyscout, 2017-12-10:
Hi,
I am trying to include python to my openwrt build. Sadly it fails.
What I have done:

```
git clone https://github.com/openwrt/openwrt.git
cd openwrt
./scripts/feeds update -a
./scripts/feeds install -a
make defconfig
```
run `make menuconfig` to select my target hardware (System: Allwinner A20, Profile: Lamobo R1) and select Language-->python-->python
You can see my config here: [gist](https://gist.github.com/nerdyscout/b89c918f2bb1fc42d7398fe32a085f82)

```
run make
```

The build log can be found here: [gist](https://gist.github.com/nerdyscout/252ae2afb50b06fb43aab1920d7f64d3)
I guess the interesting part starts somewhere in line 1378

If i don't select python the build runs smoothly, just enabled it crashes

I am running this on
Linux zbox 4.10.0-32-generic #36~16.04.1-Ubuntu SMP Wed Aug 9 09:19:02 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

I would be happy about every hint.
thx!

diizzyy, 2017-12-10:
The OpenWrt tree (master/trunk) isn't supported and you're advised to use the https://github.com/lede-project/source tree instead. This will eventually replace the current OpenWrt tree.

nerdyscout, 2017-12-14:
thx... so far I can tell this fixed my original issue. 
unfortunately I am now not able to connect to the webgui anymore... anyway, this is another problem i will try to fix somewhere else.

closed 2017-12-14 by nerdyscout