openwrt/packages #5268: Remove me as maintainer

diizzyy, 2017-12-12:
I hereby resign as maintainer for the following packages and therefore wish do be removed as maintainer.
Feel free to leave these in the tree or remove completely

```
admin/netdata
libs/confuse
net/inadyn
net/sipgrep
net/sngrep
net/stunnel
sound/fdk-aac
utils/pcsc-tools
utils/tio
```

@pprindeville 
Can you handle request this for me?

pprindeville, 2017-12-12:
@champtar How do we go about handling this?

champtar, 2017-12-12:
Hi @diizzyy 
Sorry to hear that, if you don't mind, can you explain why (I have ideas but don't want to put words in your mouth) and what we can improve?

@pprindeville I would go in the commit history of each package, see if there was commit from anyone else and offer maintainership, if we can't find anyone for any packages we will move it to package abandoned

diizzyy, 2017-12-12:
@champtar 
Thanks, I wont go into details but due to events I've decided that it's best to step away. I wish to say thanks to the package maintainers/committers and contributors who've given me help and support along the way. I don't think there's much to improve that could've changed my decision to be honest.

pprindeville, 2017-12-12:
> I don't think there's much to improve that could've changed my decision to be honest.

There's a _lot_ that could be improved, however it might not be realistic to wait for it...

diizzyy, 2018-01-09:
Closing this as I've made pull requests to arrange this myself.

closed 2018-01-09 by diizzyy

thagabe, 2018-01-10:
@diizzyy I'm sadden to see you leave. Thank you for your work and support you've given me, us, while you were part of the community!