openwrt/packages #5278: [discussion] should we remove .pyc files from all packages?

lynxis, 2017-12-14:
maintainers: @jefferyto @kissg1988 @commodo @dangowrt
(I've taken a random amount of py maintainers, nobody should feel excluded here)

Because of reproducible builds, I've noticed many (>50) python packages are unreproducible.
Those unreproducible packages are shipping **.pyc** which are unreproducible [0].

* A .pyc is a "compiled" python script file.
* Usually the .pyc got generated when the .py file got interpreted for the first time. (LEDE don't do this at runtime, only when building it via make)
* The .pyc shortcuts the startup time of python.
* .pyc can be used without the .py source file.
* .pyc will be ignored if the .py exists and .py has a different timestamp than the embedded timestamp in the .pyc

When packaging the result `python-foo.ipk` all timestamp of the files in the archive will be modified to `$SOURCE_DATE_EPOCH` [1]. So the timestamp of the .py won't match the .pyc.

What are the reasons for .pyc files in LEDE packages?
Should we remove it?
Should we fix the .pyc timestamp?

PS: I might not know all facts of the .pyc files.

[0] https://tests.reproducible-builds.org/lede/lede_ar71xx.html (search for unreproducible)
[1] https://reproducible-builds.org/specs/source-date-epoch/

commodo, 2017-12-15:
Hey,

So, the `.py` vs `.pyc` thing boils down to performance.
That is why there are now some `python-<name>-src` packages that package `.py`

There are some discussions about this:
1. this is about where it started [with me as maintainer], and `.pyc` generation was disabled : https://github.com/openwrt/packages/issues/474
2. Then I got a few reports about performance ; one to my private email saying that [on some slow SoC] a simple HelloWorld.py script runs :  `.pyc` == ~70 msecs,  `.py` == 500 msecs [ with bytecodes disabled ] ; so, then I changed Python to ship bytecodes instead of sources
2a. There is another thread regarding performance:
http://openwrt-devel.openwrt.narkive.com/2hc56Q30/openwrt-how-to-include-byte-compile-pyc-files-into-sysupgrade-image

Personal conclusions/opinions [ feel free to argue/disagree ]:
* you cannot have the best of both worlds with Python on embedded devices [micropython may be another thing] ; it would be nice to have the script files there, but then the slowness of Python interpreting them hits harder when you scale down the CPU performance
* for some reason, people like Python/Python3 [the bloat-y one] enough to run it on OpenWrt/LEDE
* I have not had complaints about Python/Python3 shipping only bytecodes ; so it seems [or I can assume] people don't care if we ship `.py`/`.pyc` as long as Python/Python3 works

I guess I got carried away with the background :)

In any case, I would choose to fix `.pyc` timestamps.
I will have to look into this, but if there are any suggestions, I'm open to them.
AFAIK: shipping only Python bytecodes is specific mostly to OpenWrt/LEDE [in all public Python packages I know of], so, it's possible that this is new territory.

But, if we fix this in the `python/python3` packages, all reproduce-ability issues should be fixed for all other python/python3 packages.
A lot [but not all] of Python/Python3 packages use build rules exported by the interpreter `python/python3` packages, which helps unify some things.

Thanks
Alex

commodo, 2017-12-27:
@lynxis 
ping

[ i know it's the holidays now ; will ping again later :) ]

lynxis, 2017-12-28:
no. chaos communication congress ;)

NeoRaider, 2017-12-28:
What do other reproducible distros do with .pyc files?

commodo, 2017-12-28:
Tbh, I don’t know.
I’d have to check.
I’ll try to do that in the upcoming days.

I suspect that the .pyc packaging is a bit unique to OpenWrt as an
open-distro.
Other distros may just ship .py files and allow .pyc generation at runtime.
Since most just target x86 and the resources are not such a big issue.

I’ll come back with some data.

commodo, 2017-12-29:
@NeoRaider 

So, for a sampling of a few distros, see below:

1. FreeBSD:
* https://tests.reproducible-builds.org/freebsd/freebsd.html  - does not seem to have Python in the list ; or I cannot find it there
* but looking through this: pkg.freebsd.org/freebsd:12:x86:64/latest/All/python27-2.7.14_1.txz  ; it does look like Python bytecodes are shipped with source ; AFAICT, the timestamp inside the file may not be consistent, but I could be wrong

2. Fedora, seems to be behind on this : https://tests.reproducible-builds.org/rpms/fedora-23.html
I saw that 2016 March was the last time it was run, and many python packages were not reproducible; though I did not see any Python interpreter package there.

3. Debian: https://tests.reproducible-builds.org/debian/reproducible.html
Seems to be at the forefront of reproducible-ality.
* The list of unreproducible builds: https://tests.reproducible-builds.org/debian/unstable/amd64/index_no_notes.html
* Python 2.7 seems unreproduce-able at binary level : https://tests.reproducible-builds.org/debian/rb-pkg/unstable/amd64/python2.7.html
* After opening up the .deb file here: http://ftp.us.debian.org/debian/pool/main/p/python2.7/libpython2.7-stdlib_2.7.14-4_amd64.deb  ; I did not find any .pyc / .pyo files ; so, on Debian,I'd assume that's how it stays reproducible

4. Arch Linux: https://tests.reproducible-builds.org/archlinux/archlinux.html
* Seems build is failed for Python & Python2 with ` failed to build while running tests ` ; maybe if I go back and re-check it could be passing
* I found the packaging rules  here tho: https://git.archlinux.org/svntogit/packages.git/tree/trunk/PKGBUILD?h=packages/python
* I'd assume that Arch is not reproducible either, since I'd assume that Arch does not disable bytecode compilation in the `libinstall` rule as OpenWrt does

All in all, I am not sure if other distros have got to care about Python bytecodes, so we could be in the lead with this, and we could push this for consideration to the Python project.

If this goes in I'll re-check the link that @lynxis sent and if all is good I can sent this upstream for consideration.

commodo, 2018-01-03:
Hey,

An update ; seems I forgot a bit about any potential discussions within Python [upstream] about reproduce-ability.

Yesterday, I checked and found this [and joined in]:  https://bugs.python.org/issue29708

Seems that [PEP-552](https://www.python.org/dev/peps/pep-0552/) was created to address this officially.
It's now in Python 3.7.
No idea yet about other Python versions [specifically 2.7].

Will try to keep track of that until it settles.

commodo, 2018-01-06:
More stuff for reproduce-ability:
https://github.com/openwrt/packages/pull/5360

commodo, 2018-01-19:
@lynxis 
is there a way to re-kick the build here sooner:
https://tests.reproducible-builds.org/lede/lede_ar71xx.html

commodo, 2018-03-14:
@lynxis 

seems the job has stopped running ; last run was on the ~20th of January ;
https://tests.reproducible-builds.org/lede/lede_ar71xx.html

any thoughts on when it would be back ?

commodo, 2018-03-14:
fwiw: i've added a few changes that may help Python3 be reproducible and i'd be curios what the build says

neheb, 2020-01-12:
What's the progress on this?

commodo, 2020-01-13:
So, there was a build that also included packages feed.
I can't seem to find it.

I found this:
https://tests.reproducible-builds.org/openwrt/openwrt.html

As far as I remember, there was some work that I did to make Py3 more reproducible and I think the results were pretty good.
But I am not sure now how many packages were left to be unreproducible.

It would be great to have a view regarding all packages, where we can see this.
Then we can have a better answer.

@lynxis 
Any thoughts?

jefferyto, 2020-01-14:
Current tests on reproducible-builds.org are building base/core packages only. Building all packages was ["temporarily disabled"](https://salsa.debian.org/qa/jenkins.debian.net/commit/5f5b2c29f973160a0ddbeea9b342a749c657be88) in Oct 2018, [re-enabled](https://salsa.debian.org/qa/jenkins.debian.net/commit/13e62b27a0cff8af478d3c50e2dec9a35da68071) then [disabled again](https://salsa.debian.org/qa/jenkins.debian.net/commit/cbfb445599411eab5c972c860856a251760fe423) on the same day in Feb 2019. I assume there are problems building all packages but there are no details in the commit messages.

commodo, 2023-05-27:
this seems to have died 

closing;

feel free to re-open if i'm wrong

closed 2023-05-27 by commodo