openwrt/packages #5289: shairport-sync lagging behind

nos1609, 2017-12-17:
Maintainer: @thess @mikebrady 

As far as I can see the package is very outdated comparing to the master repo. Awaiting v3.1.5.

Thanks for keeping it alive.

thess, 2017-12-18:
You are right - It needs updating...

mikebrady, 2017-12-18:
Thanks for the reminders— I have neglected it, I’m afraid. I’ll do some work on it over the Christmas break.

thess, 2017-12-18:
@mikebrady - I've got 3.1.6 ready to go. If you want to work on something, the ALAC library would be a nice addition. It kind of fell flat last attempt.

mikebrady, 2017-12-18:
Thanks Ted. I will have a look alright. It fell flat on licensing grounds, but also someone else seems to have done a better job. I'll investigate.

thess, 2018-01-08:
Resolved by https://github.com/openwrt/packages/commit/54927ab1dff3b54d10c5f95b33f224c8f604ef8f.
closing this

closed 2018-01-08 by thess