openwrt/packages #5293: curl: https request returns 'Illegal instruction'

ogarcia, 2017-12-18:
Environment:

```
root@LEDE:~# cat /etc/os-release
NAME="LEDE"
VERSION="17.01.4, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot 17.01.4"
VERSION_ID="17.01.4"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r3560-79f57e422d"
LEDE_BOARD="mpc85xx/generic"
LEDE_ARCH="powerpc_8540"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot 17.01.4 r3560-79f57e422d"
```

Description:

I'm using oficial stable release for _TP-Link TL-WDR4900 v1_ and when I try to use `curl` I give the following error:
```
root@LEDE:~# curl https://google.com
Illegal instruction
```

I have installed all curl dependencies and ca-certificates:
```
root@LEDE:~# opkg list-installed | egrep "ssl|curl|mbed|ddns"
curl - 7.52.1-6
ddns-scripts - 2.7.6-13
ddns-scripts_cloudflare.com-v4 - 2.7.6-13
libcurl - 7.52.1-6
libmbedtls - 2.6.0-1
libopenssl - 1.0.2n-1
luci-app-ddns - 2.4.8-2
openvpn-openssl - 2.4.4-2
```
And wget works well:
```
root@LEDE:~# wget https://google.com
--2017-12-18 13:25:10--  https://google.com/
Resolving google.com... 178.60.128.37, 178.60.128.42, 178.60.128.20, ...
Connecting to google.com|178.60.128.37|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://www.google.es/?gfe_rd=cr&dcr=0&ei=prM3Wp3qOeGs8wfNpI_gDg [following]
--2017-12-18 13:25:10--  https://www.google.es/?gfe_rd=cr&dcr=0&ei=prM3Wp3qOeGs8wfNpI_gDg
Resolving www.google.es... 172.217.21.67, 2a00:1450:4006:807::2003
Connecting to www.google.es|172.217.21.67|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'index.html'

index.html                                       [ <=>                                                                                          ]  12.82K  81.4KB/s    in 0.2s    

2017-12-18 13:25:11 (81.4 KB/s) - 'index.html' saved [13132]
```

There is an issue with `curl - 7.52.1-6`?

Marz1pan, 2018-01-03:
Hi there,

> There is an issue with curl - 7.52.1-6?
No  is is an issue with ca-certificates !

i ran into the same trouble using curl over https :-(
It took me a while to discover that curl is missing the file:     /etc/ssl/certs/ca-certificates.crt
Unfortunatly it is not included in the ca-certificates package :-(

My "quick and dirty" workaround was that i copied the full content of
/etc/ssl/certs/ from my Desktop PC (running Linux Mint) straight ahead to the openwrt router.
I allready send an email to the ca-certificates maintainer, but he never dared to answer me :-(

Greetings

ogarcia, 2018-01-03:
@Marz1pan your issue not it the same, in your case you must install `ca-bundle` package. The file `/etc/ssl/certs/ca-certificates.crt` is no more included into `ca-certificates`.

zenitraM, 2018-02-12:
I just stumbled upon the exact same issue with my TL-WDR4900 - might be an issue with the mpc85xx build?

diizzyy, 2018-07-05:
@zenitraM @ogarcia 
Can you guys give 18.X a try and/or trunk/master?

diizzyy, 2018-08-15:
@hnyman 
Close due to reporter timeout

ogarcia, 2018-08-15:
@diizzyy @hnyman Please not close it. I cannot test 18.X until september because my TL-WDR4900 is in other location. Please wait a few days and I'll give you feedback.

diizzyy, 2018-08-15:
@ogarcia 
It's still the wrong repo, this should be filed here: https://bugs.openwrt.org

dibdot, 2018-08-17:
@ogarcia if this issue still still exsist for you, please use the main project bug tracker as @diizzyy said.

closed 2018-08-17 by dibdot

PolynomialDivision, 2022-10-15:
@ogarcia Did you report that upstream?