openwrt/packages #5304: dnscrypt-proxy: dnscrypt-resolvers outdated

EnergyMouse, 2017-12-20:
Maintainer: @dibdot 
Environment: OpenWrt SNAPSHOT, r5578-f704b643b9

Description: dnscrypt-resolvers (part of dnscrypt-proxy) has not been updated for a long time, contains non-working and miss new resolvers. Also dnscrypt-resolvers was moved to separate repository https://github.com/DNSCrypt/dnscrypt-resolvers .

dibdot, 2017-12-20:
I'm not the maintainer, I've just written a LuCI frontend for this package! @damianorenfer 
BTW, you can use the luci frontend to update your local resolver list with just one click.
A more stable download link is located here: https://download.dnscrypt.org/dnscrypt-proxy/dnscrypt-resolvers.csv

thess, 2018-01-08:
Package update pending (#5381)  - will close this when merged. Waiting for ACK from @damianorenfer.

closed 2018-01-08 by thess