openwrt/packages #5306: netdata: openwrt compilation error

sonamukh, 2017-12-20:
-------------------------------

Maintainer: @<github-user> (find it by checking history of the package Makefile)
Environment: (OpenWRT, netdata pacakge)

Description:

Hi,
  I am trying to complie netdata package within openwrt and running into the following error:

net_snmp6.o proc_net_softnet_stat.o proc_net_stat_conntrack.o proc_net_stat_synproxy.o zfs_common.o proc_spl_kstat_zfs.o proc_stat.o proc_sys_  kernel_random_entropy_avail.o proc_vmstat.o proc_uptime.o sys_kernel_mm_ksm.o sys_fs_cgroup.o -lm  -L/home/centos/openwrt/staging_dir/target-mi  ps_34kc_uClibc-0.9.33.2/usr/lib -lz -L/home/centos/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -luuid
statsd.o: In function `statsd_parse_float.constprop.9':
statsd.c:(.text.unlikely.statsd_parse_float.constprop.9+0xf6): undefined reference to `powl'
statsd.c:(.text.unlikely.statsd_parse_float.constprop.9+0x148): undefined reference to `powl'
statsd.o: In function `statsd_process_counter':
statsd.c:(.text.unlikely.statsd_process_counter+0xce): undefined reference to `llrintl'
statsd.o: In function `statsd_flush_timer_or_histogram':
statsd.c:(.text.statsd_flush_timer_or_histogram+0x52): undefined reference to `roundl'
statsd.c:(.text.statsd_flush_timer_or_histogram+0x8e): undefined reference to `roundl'
statsd.c:(.text.statsd_flush_timer_or_histogram+0xc8): undefined reference to `roundl'
statsd.c:(.text.statsd_flush_timer_or_histogram+0x102): undefined reference to `roundl'
statsd.c:(.text.statsd_flush_timer_or_histogram+0x13c): undefined reference to `roundl'
statsd.o:statsd.c:(.text.statsd_flush_timer_or_histogram+0x176): more undefined references to `roundl' follow
appconfig.o: In function `appconfig_get_float':
appconfig.c:(.text.appconfig_get_float+0x102): undefined reference to `powl'
appconfig.c:(.text.appconfig_get_float+0x158): undefined reference to `powl'
eval.o: In function `parse_one_full_operand':
eval.c:(.text.parse_one_full_operand+0x2ca): undefined reference to `powl'
eval.c:(.text.parse_one_full_operand+0x324): undefined reference to `powl'
health_config.o: In function `str2ld':
health_config.c:(.text.str2ld+0x106): undefined reference to `powl'
health_config.o:health_config.c:(.text.str2ld+0x15a): more undefined references to `powl' follow
rrd2json.o: In function `rrd_stats_api_v1_charts_allmetrics_shell':
rrd2json.c:(.text.rrd_stats_api_v1_charts_allmetrics_shell+0xd0): undefined reference to `roundl'
rrd2json.c:(.text.rrd_stats_api_v1_charts_allmetrics_shell+0x114): undefined reference to `roundl'
rrd2json.c:(.text.rrd_stats_api_v1_charts_allmetrics_shell+0x198): undefined reference to `roundl'
statistical.o: In function `standard_deviation':
statistical.c:(.text.standard_deviation+0xea): undefined reference to `sqrtl'
storage_number.o: In function `print_calculated_number':
storage_number.c:(.text.print_calculated_number+0x32): undefined reference to `modfl'
storage_number.c:(.text.print_calculated_number+0xa8): undefined reference to `llrintl'
unit_test.o: In function `str2ld':
unit_test.c:(.text.str2ld+0x106): undefined reference to `powl'
unit_test.c:(.text.str2ld+0x15a): undefined reference to `powl'
unit_test.o: In function `run_test':
unit_test.c:(.text.run_test+0x386): undefined reference to `roundl'
unit_test.c:(.text.run_test+0x39e): undefined reference to `roundl'
unit_test.c:(.text.run_test+0x452): undefined reference to `roundl'
unit_test.c:(.text.run_test+0x46a): undefined reference to `roundl'
collect2: error: ld returned 1 exit status

It looks like the linker flag "-lm" is not set properly in openwrt. Any help is appreciated.

Thanks,

krupatil, 2018-01-24:
Hi, did you ever find any solution to this?

thagabe, 2018-01-25:
Just built netdata yesterday, everything is working great

krupatil, 2018-01-25:
How did you fix undefined reference to roundl?

thagabe, 2018-01-25:
have you tried cleaning your build folder? make dirclean and rebuild

krupatil, 2018-01-25:
Yes, I did.

thagabe, 2018-01-25:
What are you building it for? (target)

krupatil, 2018-01-25:
Arm target.

thagabe, 2018-01-25:
do you have any patches applied out of tree

diizzyy, 2018-04-01:
I highly doubt anyone cares about OpenWrt 14.07 since that (according to Google) seems to be the version that uses uClibc 0.9.33.2

diizzyy, 2018-07-05:
@hnyman @champtar 
Please close, not a valid bug report (outdated toolchain etc)

closed 2018-07-05 by yousong