openwrt/packages #5310: ptyhon-2.7.14: Compile Python 2.7.14 error macOS 10.13.2

kholisrag, 2017-12-21:
Maintainer: @commodo 
Environment: LEDE
Package: python2
Description:
Compile for python 2.7.14 error macOS 10.13.2 High Sierra

`(cd /Volumes/Lede/Lede/build_dir/hostpkg/Python-2.7.14/; if [ -x configure ]; then cp -fpR /Volumes/Lede/Lede/scripts/config.{guess,sub} /Volumes/Lede/Lede/build_dir/hostpkg/Python-2.7.14// && CC="gcc" CFLAGS="-O2 -I/Volumes/Lede/Lede/staging_dir/host/include -I/Volumes/Lede/Lede/staging_dir/host/usr/include -I/Volumes/Lede/Lede/staging_dir/hostpkg/include -I/Volumes/Lede/Lede/staging_dir/target-mips_24kc_musl/host/include" CPPFLAGS="-I/Volumes/Lede/Lede/staging_dir/host/include -I/Volumes/Lede/Lede/staging_dir/host/usr/include -I/Volumes/Lede/Lede/staging_dir/hostpkg/include -I/Volumes/Lede/Lede/staging_dir/target-mips_24kc_musl/host/include" LDFLAGS="-L/Volumes/Lede/Lede/staging_dir/host/lib -L/Volumes/Lede/Lede/staging_dir/host/usr/lib -L/Volumes/Lede/Lede/staging_dir/hostpkg/lib -L/Volumes/Lede/Lede/staging_dir/target-mips_24kc_musl/host/lib -Wl,--no-as-needed -lrt $(pkg-config --static --libs libcrypto libssl)" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-apple-darwin17.3.0 --host=x86_64-apple-darwin17.3.0 --build=x86_64-apple-darwin17.3.0 --program-prefix="" --program-suffix="" --prefix=/Volumes/Lede/Lede/staging_dir/hostpkg --exec-prefix=/Volumes/Lede/Lede/staging_dir/hostpkg --sysconfdir=/Volumes/Lede/Lede/staging_dir/hostpkg/etc --localstatedir=/Volumes/Lede/Lede/staging_dir/hostpkg/var --sbindir=/Volumes/Lede/Lede/staging_dir/hostpkg/bin --without-cxx-main --without-pymalloc --with-threads --prefix=/Volumes/Lede/Lede/staging_dir/hostpkg --exec-prefix=/Volumes/Lede/Lede/staging_dir/hostpkg --with-system-expat=/Volumes/Lede/Lede/staging_dir/hostpkg --with-system-ffi=no --with-ensurepip=install CONFIG_SITE= ; fi )
checking build system type... x86_64-apple-darwin17.3.0
checking host system type... x86_64-apple-darwin17.3.0
checking for python2.7... python2.7
checking for --enable-universalsdk... no
checking for --with-universal-archs... 32-bit
checking MACHDEP... darwin
checking EXTRAPLATDIR... $(PLATMACDIRS)
checking for --without-gcc... no
checking for --with-icc... no
checking for x86_64-apple-darwin17.3.0-gcc... gcc
checking whether the C compiler works... no
configure: error: in `/Volumes/Lede/Lede/build_dir/hostpkg/Python-2.7.14':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [/Volumes/Lede/Lede/build_dir/hostpkg/Python-2.7.14/.configured] Error 77
make[3]: Leaving directory `/Volumes/Lede/Lede/feeds/packages/lang/python/python'
make[2]: *** [package/feeds/packages/python/python/host/compile] Error 2
make[2]: Leaving directory `/Volumes/Lede/Lede'
make[1]: *** [/Volumes/Lede/Lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/Volumes/Lede/Lede'
make: *** [world] Error 2`

this is the log for complete detail...

[config.log](https://github.com/openwrt/packages/files/1579195/config.log)

commodo, 2017-12-21:
I wish this report would have come a few months ago when I still had a Mac :)

I'll try to guess via the `config.log` output, but the overall building-on-a-Mac is problematic after each version that gets released.
And homebrew also changes from time to time.
Last I checked was Sierra.

I'll probably try to find a solution [MacOS X on a VM, or something] to take a look.

kholisrag, 2017-12-21:
That's Fine...
I just can build Ubuntu in VM too...
It's nothing major for me...
Thanks for your hard work...

jpc, 2018-02-20:
This was caused by 1c54e2b0fbe2e9a7dfb050b45ff79a0f72b8582f which added `-Wl,--no-as-needed` to build flags on all platforms. This flag is not supported by the macOS linker.
Reverting that commit helps a bit but unfortunately I have further problems with the _ctypes module: #5638

commodo, 2018-02-20:
that flag ( `-Wl,--no-as-needed` ) was added to support Ubuntu 12.04 ; which is one of those distros people like to keep around forever ;
maybe some sort condition could be added to check for build-host and add it on Linux ; 

regarding ctypes ; will move discussion to other issue

commodo, 2018-02-20:
maybe we could do
```
ifneq ($(HOST_OS),Darwin)
 HOST_LDFLAGS += \
	-Wl,--no-as-needed -lrt
endif 
```

i admit it would be saner/better to check for linker, but that seems a bit more difficult atm ;

commodo, 2018-03-14:
@kholisrag 
can you please test with the changes in PR https://github.com/openwrt/packages/pull/5758 ?

thanks
Alex

diizzyy, 2018-08-15:
@kholisrag 
Ping