openwrt/packages #5320: Remove lua and luci-base dependencies in luci-app-sqm

eterry1388, 2017-12-24:
This is causing `recursive dependency detected!` error when there are other packages that depend on `lua`.

https://github.com/openwrt/packages/blob/8c69bc44364441157ecf61707bc7ce4c74ea4746/net/sqm-scripts/Makefile#L54

eterry1388, 2017-12-24:
See #3620 for a suggestion to remove these dependencies from this package (scroll down quite a ways to find the comment by @hnyman

diizzyy, 2019-03-28:
Wrong repo, closing

karlp, 2019-03-28:
@diizzyy  what do you mean, wrong repo?  THe luci app is provided in the same package makefile, which is in this package.

karlp, 2019-03-28:
I still feel that this is a bug in your "mypackage" deps though, there's nothign wrong with having other packages depend on lua.

diizzyy, 2019-03-28:
Terribly sorry, I misread the url

tohojo, 2020-01-17:
Hmm, don't mind removing the lua dependency. And since we're depending on luci-compat now, that will also pull in luci-base, so I guess we can remove that as well (at least until I figure out how to avoid the -compat dependency).

tohojo, 2020-01-17:
BTW, thanks @neheb for assigning these bugs to me; I don't follow the repo issues so if I'm not tagged in a report I generally don't see it (as you can probably tell)... :)

commit b59f7eb8 (2020-01-17): luci-app-sqm: Remove lua and luci-base dependencies

We depend on luci-compat now, which will pull in luci-base, which in turn
will pull in lua. So remove those two dependencies to maybe help with
recursive dependency resolution errors.

Fixes #5320.
closed 2020-01-17 by tohojo