openwrt/packages #5321: How to install from fork?

innovate-invent, 2017-12-24:
Hey guys,
I am new to openwrt and I added some functionality to the tinyproxy package.
Before submitting a pull request I want to test my changes.
I forked this repo but I am not clear how to have opkg install from the fork.
The opkg documentation I found refers to ipk files and I am confused how this repo ties in.

As a side, can anyone tell me what script calls the makefiles in each package?
I need to hunt down where CONFIGURE_ARGS is used in the makefile.

Thanks

hnyman, 2017-12-28:
You need to compile the package at your local build system (of the local copy of the fork), then upload the compiled ipk file to the router and opkg install it.

> can anyone tell me what script calls the makefiles in each package?
> I need to hunt down where CONFIGURE_ARGS is used in the makefile.

Default configure action is here (and the defaults are defined just above that):
https://github.com/lede-project/source/blob/master/include/package-defaults.mk#L114

hnyman, 2017-12-28:
This issue is better suited for forum, so I will close this.

closed 2017-12-28 by hnyman