openwrt/packages #5322: exfat-nofuse: update needed if lede switch to kernel 4.14

DazzyWalkman, 2017-12-25:
I am trying to compile lede x86 trunk r5616 with kernel 4.14.6 selected.
The exfat-nofuse has to be updated to include following commit:
https://github.com/dorimanx/exfat-nofuse/commit/1e91a9feb0e45161d3c02f9257a9b6c44b9e2739
otherwise compilation would fail with 'CURRENT_TIME_SEC undeclared'

I successfully tested compile exfat-nofuse dorimanx tree latest git 2017-06-20 
 de4c760bc9a05ead83bc3ec6eec6cf1fb106f523.
Look good to me.

bouwew, 2018-01-07:
I ran into the same problem when trying to compile for R7800 with kernel 4.14.
So, I support this request.

yousong, 2018-01-09:
ping exfat-nofuse's current maintainer @br101

br101, 2018-01-09:
Hi! Yes, it makes sense to update to the latest version! 
BTW: Anyone willing to take over maintainership? I have no project using this code any more.

diizzyy, 2018-01-09:
https://github.com/openwrt/packages/pull/5388

diizzyy, 2018-01-21:
@br101 
Just submit a pr removing yourself from the maintainer line, that seems to be easiest way.

yousong, 2018-01-23:
@diizzyy   Do you feel like proposing a change to the build system to

 - download the auto-generated tarball
 - unpack it
 - repack it in a reproducible way

This way we save bandwidth and will have predictable hash values.

diizzyy, 2018-01-23:
You should really refrain to do unnecessary I/O if you want somewhat decent scaling and since you're deadset on not using the auto-generated tarballs for whatever reason I think the proposed solution is a step backward. Relying on auto-generated tarballs may be unofficial but it does work just fine in ~99% of all cases and speeds up operations quite a bit especially if you want apply it tree-wide.

br101, 2018-01-23:
Hi! I suppose this discussion is not specific to exfat and rather something that should be discussed OpenWRT/LEDE wide. For now it should be OK to follow the common practice of other packages. 

And when you make a PR can you please also remove me as maintainer? That would be the most efficient way to handle that.

yousong, 2018-01-24:
@br101 , @diizzyy , I just opened pull request #5494 .  Let's move on to fix the issue at hand first.

commit 780330a2 (2018-01-25): exfat-nofuse: bump to source date 2017-06-20

Fixes #5322, #5444.  Closes #5388
closed 2018-01-25 by yousong