openwrt/packages #5326: i2pd: Improve the package

LLE8, 2017-12-27:
@yangfl 
LEDE 17.01.4

I think, this build can be improved.
Look at https://forum.openwrt.org/viewtopic.php?pid=369287#p369287 please
and above, https://forum.openwrt.org/viewtopic.php?pid=368965#p368965
and https://forum.openwrt.org/viewtopic.php?pid=368966#p368966

Dependence on boost-regex is not needed, i think (~850 KB).
Do not forget add home dir creation in init.
If you wrote `MAKE_FLAGS+=USE_AESNI=no USE_AVX=no` in Makefile, you may delete patch with this options.
Add pidfile creation in init with correct permissions.
May be add tuning of tcp/ip in init (sysctl), i am not sure ,

LLE8, 2017-12-27:
And add i2pd user creation in Makefile, do not run as root.

yangfl, 2017-12-27:
@dartraiden Instead, I'm considering explicitly moving working dir into /etc/i2pd, since peer info can take up to 100MB which is quite large compared with typical router memory. So I'd suggest using extroot.

LLE8, 2017-12-27:
As @orignal said, only netDB and addresses.csv are nessesary. It is less then 10 MB. But it is permanently rewritten. Flash will die.

commit 1a0801ff (2018-01-06): Merge pull request #5327 from yangfl/master

i2pd: Improve the package, fix #5326
commit fba9ee15 (2018-01-06): i2pd: Improve the package, fix #5326
closed 2018-01-06 by hnyman