openwrt/packages #5329: bird4/6 - Package should add the TCP_MD5SIG kmod?

Strykar, 2017-12-28:
_RFC2385 specifies a method of giving MD5 protection to TCP sessions. Its main (only?) use is to protect BGP sessions between core routers on the Internet_. TCP-MD5 (`CONFIG_TCP_MD5SIG`) is a common method of BGP inter-router authentication. bird appears to have support for this, but openwrt/lede do not create a kernel module available as a binary download via opkg.

Is this the bird package issue or an openwrt one?

eloicaso, 2018-01-03:
Ping @Noltari.

From my perspective, if this kmod would be required, it should rely on the daemon's dependencies and not on the configuration wrapper ones. I will investigate the issue in more detail once I am back home from holidays with my computer.

Strykar, 2019-01-29:
Hi @eloicaso Any updates from you or @Noltari?

Strykar, 2020-12-28:
Wireguard is a better solution now, closing.

closed 2020-12-28 by Strykar