openwrt/packages #5333: PHP causing problem

ibnbd, 2017-12-29:
@mhei thank you for updating PHP version, however I can't `make` my code because of PHP since the update.

```

In file included from /home/ahstanin/source/build_dir/target-mipsel_24kc_musl/php-7.2.0/ext/pcre/pcrelib/sljit/sljitLir.c:1747:0,
                 from /home/ahstanin/source/build_dir/target-mipsel_24kc_musl/php-7.2.0/ext/pcre/pcrelib/pcre_jit_compile.c:62:
/home/ahstanin/source/build_dir/target-mipsel_24kc_musl/php-7.2.0/ext/pcre/pcrelib/sljit/sljitNativeMIPS_common.c: In function 'sljit_has_cpu_feature':
/home/ahstanin/source/build_dir/target-mipsel_24kc_musl/php-7.2.0/ext/pcre/pcrelib/sljit/sljitNativeMIPS_common.c:506:3: error: a label can only be part of a statement and a declaration is not a statement
   sljit_sw fir;
   ^
Makefile:589: recipe for target 'ext/pcre/pcrelib/pcre_jit_compile.lo' failed
make[4]: *** [ext/pcre/pcrelib/pcre_jit_compile.lo] Error 1
make[4]: Leaving directory '/home/ahstanin/source/build_dir/target-mipsel_24kc_musl/php-7.2.0'
Makefile:579: recipe for target '/home/ahstanin/source/build_dir/target-mipsel_24kc_musl/php-7.2.0/.built' failed
make[3]: *** [/home/ahstanin/source/build_dir/target-mipsel_24kc_musl/php-7.2.0/.built] Error 2
make[3]: Leaving directory '/home/ahstanin/source/feeds/packages/lang/php7'
package/Makefile:109: recipe for target 'package/feeds/packages/php7/compile' failed
make[2]: *** [package/feeds/packages/php7/compile] Error 2
make[2]: Leaving directory '/home/ahstanin/source'

```

pprindeville, 2017-12-30:
Please pick better titles in the future.

In any case, this is fixed with PR #5335.

closed 2017-12-30 by pprindeville

pprindeville, 2017-12-31:
Did this fix your issue?  Please confirm.

ibnbd, 2018-01-01:
@pprindeville thank you for your update, actually I patched it on your PR.