openwrt/packages #5339: Please add module SARL

DocMAX, 2017-12-31:
Could you please add this module?

```
Can't locate Authen/SASL.pm in @INC (you may need to install the Authen::SASL module) (@INC contains: /usr/lib/perl5/5.22 .) at sendxmpp line 20.
BEGIN failed--compilation aborted at sendxmpp line 20.
```

pprindeville, 2017-12-31:
It's not enough to have `Authen::SASL` -- it's just a framework.  What underlying plug-ins do you. Please see [Authen::SASL Description](http://search.cpan.org/~gbarr/Authen-SASL-2.16/lib/Authen/SASL.pod#DESCRIPTION) need?

pprindeville, 2018-01-02:
If you can submit a patch to package `Authen::SASL::XS` I'll include it.

pprindeville, 2018-02-13:
@DocMAX See PR #5612.  If you can test it and verify it works, I can merge it.

closed 2018-04-02 by yousong