openwrt/packages #5341: MWAN3: Stops NoDogSplash from working

bluewavenet, 2017-12-31:
Please make sure that the issue subject starts with `<package-name>: ` so that it's easily identifiable.

This repo here is ONLY for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt: https://dev.openwrt.org/newticket
 - LEDE: https://bugs.lede-project.org/
 - Most LuCI packages: https://github.com/openwrt/luci/issues

Thanks for your contribution
Please remove this text (before ---) and fill the following template
-------------------------------

Maintainer: @<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWRT/LEDE version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

feckert, 2018-01-08:
@bluewavenet 
On master there was added the possibility to make the mark flags for mwan3 configurable. Have you tested this?

bluewavenet, 2018-01-08:
@feckert
I've not seen that. Will give it a try.

bluewavenet, 2018-01-08:
@feckert 
I don't see that anywhere. Any pointers to where? ;-)

bluewavenet, 2018-01-08:
@feckert 
Ah! Is it MMX_MASK in mwan3.sh?

feckert, 2018-01-08:
@bluewavenet 
yes this is the right config option.
For now it is set to 0xFF which gives you the possibility to administrate 1-250 interfaces (see openwrt doku 250 is the max because of iproute2)
253 (blackhole)
254 (unreachable)
255 (default)

you could set this to 0xAA this will divide the interface by 16 and you will have now 16 Intefaces to administrate. The omitted marks could be used for example by NDS.

bluewavenet, 2018-01-08:
@feckert 
mwan3.sh seems to be setting to 0xff00, not 0x00ff, am I reading this correctly?
Lines 20,21:
config_load mwan3
config_get MMX_MASK globals mmx_mask '0xff00'

I won't have suitable hardware for testing for a few days, but will do some testing then.

feckert, 2018-01-08:
@bluewavenet 
yes this is right it uses 0xff00 if nothing is set
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L21

bluewavenet, 2018-01-08:
@feckert 
and not 0xff.
Version 1 used 0xff ?
This might explain the problem with Nodogsplash.
I need to do some testing in the light of this.
Thanks :-)

feckert, 2018-03-15:
@bluewavenet is this issue solved?

bluewavenet, 2018-03-15:
@feckert 
I have not had time to look at it so mo progress but I think it is relatively straightforward to fix. I will have to make time asap and revisit ;-)

bluewavenet, 2018-03-27:
@feckert 
It is now working on 2.6.11-2 (once I got it started ;-) ), by setting higher bits than mwan3 is using eg 0x30000 etc for Nodogsplash. Running VERY well - thanks for all the work put into it!

feckert, 2018-04-03:
@bluewavenet i think we could close this issue?

bluewavenet, 2018-04-03:
@feckert - indeed ;-)

closed 2018-04-03 by bluewavenet