openwrt/packages #5343: lighttpd 1.4.48 - no openssl module

silvah2, 2018-01-02:
there is no mod-openssl which is now required to run lighttpd

pprindeville, 2018-01-02:
I'm not sure I understand what you're saying.  I set the following:

```
%  grep -e libopenssl -e php7 .config | grep -v '^#'
CONFIG_PACKAGE_php7=y
CONFIG_PACKAGE_php7-mod-openssl=y
CONFIG_PACKAGE_php7-mod-snmp=y
CONFIG_PACKAGE_libopenssl=y
% 
```

and then built:

```
% make -j1 package/php7/{clean,compile} V=s
% pushd build_dir/target-x86_64_musl/php-7.2.0/ipkg-x86_64
% ls -Rl php7-mod-openssl
php7-mod-openssl:
total 12
drwxr-xr-x 2 philipp philipp 4096 Jan  2 12:30 CONTROL/
drwxr-xr-x 3 philipp philipp 4096 Jan  2 12:30 etc/
drwxr-xr-x 3 philipp philipp 4096 Jan  2 12:30 usr/

php7-mod-openssl/CONTROL:
total 12
-rw-r--r-- 1 philipp philipp 289 Jan  2 12:30 control
-rwxr-xr-x 1 philipp philipp 160 Jan  2 12:30 postinst*
-rwxr-xr-x 1 philipp philipp 117 Jan  2 12:30 prerm*

php7-mod-openssl/etc:
total 4
drwxr-xr-x 2 philipp philipp 4096 Jan  2 12:30 php7/

php7-mod-openssl/etc/php7:
total 4
-rw-r--r-- 1 philipp philipp 21 Jan  2 12:30 20_openssl.ini

php7-mod-openssl/usr:
total 4
drwxr-xr-x 3 philipp philipp 4096 Jan  2 12:30 lib/

php7-mod-openssl/usr/lib:
total 4
drwxr-xr-x 2 philipp philipp 4096 Jan  2 12:30 php/

php7-mod-openssl/usr/lib/php:
total 140
-rwxr-xr-x 1 philipp philipp 139555 Jan  2 12:30 openssl.so*
% 
```

Did you select `libopenssl` first?

silvah2, 2018-01-02:
I' talking about lighttpd 1.4.48 module, not php7.

 lighttpd -f /etc/lighttpd/lighttpd.conf -tt
2018-01-02 20:40:24: (plugin.c.229) dlopen() failed for: /usr/lib/lighttpd/mod_openssl.so Error loading shared library /usr/lib/lighttpd/mod_openssl.so: No such file or directory
2018-01-02 20:40:24: (server.c.1141) loading plugins finally failed

silvah2, 2018-01-02:
I'm talking about lighttpd module

 lighttpd -f /etc/lighttpd/lighttpd.conf -tt
2018-01-02 20:46:36: (configfile.c.60) Warning: please add "mod_openssl" to server.modules list in lighttpd.conf.  A future release of lighttpd 1.4.x *will not* automatically load mod_openssl and lighttpd *will not* use SSL/TLS where your lighttpd.conf contains ssl.* directives
2018-01-02 20:46:36: (plugin.c.229) dlopen() failed for: /usr/lib/lighttpd/mod_openssl.so Error relocating /usr/lib/lighttpd/mod_openssl.so: convert_to_long: symbol not found
2018-01-02 20:46:36: (server.c.1141) loading plugins finally failed

pprindeville, 2018-01-02:
> I' talking about lighttpd 1.4.48 module, not php7.

You're right.  Sorry.  Both kids have colds right now and I've not really slept in 5 days.

pprindeville, 2018-01-03:
@silvah2 Please cherry-pick and test fix.

closed 2018-01-03 by pprindeville

bluewavenet, 2018-05-06:
@pprindeville 
I have just started to get this problem on my automated imagebuilder (17.01.04).
Prior to April the 25th 2018 I was getting lighttpd version 1.4.45-3 
but after 25th April I am getting version 1.4.48-1
If anything was pushed into the release it should have been 1.4.48-2 with this fix?

bluewavenet, 2018-05-06:
lighttpd is currently failing to compile in snapshots too.