openwrt/packages #5347: dnscrypt-proxy - not working (flock erreor)

build000, 2018-01-03:
OpenWrt SNAPSHOT r5671-c5ca1c9 / LuCI Master (git-18.003.32380-b121f95) 
```
Wed Jan  3 19:14:53 2018 daemon.notice procd: /etc/init.d/dnscrypt-proxy: /etc/rc.common: /lib/functions/procd.sh: line 512: flock: not found
Wed Jan  3 19:14:53 2018 user.notice root: warning: procd flock for dnscrypt-proxy failed

dibdot, 2018-01-03:
`/lib/functions/procd.sh: line 512: flock: not found`
Please make sure, that your busybox binary includes flock support. This has been added a few days ago:

https://git.lede-project.org/?p=openwrt/openwrt.git;a=commit;h=2277cd1249a6952228f36b768beea117c91123a0

build000, 2018-01-03:
@dibdot 
Actually, I had to overlook it because I had to manually modify this file, in which I added some functions. After adding the "flock" function, everything works without this error.
Problem solved - I am closing the issue.

closed 2018-01-03 by build000

daniel-santos, 2018-10-19:
hrmm. Shouldn't this be auto-selected by whatever triggers its use?