openwrt/packages #5350: imagemagick: support new file formats (BPG, WEBP)

eliz82, 2018-01-04:
Maintainer: no idea, i did not find the package in github repo https://github.com/openwrt/packages/tree/master/multimedia ???
Environment: Entware 3x, Voxel Firmware, Netgear R7800

Description:
`opkg install imagemagick imagemagick-jpeg imagemagick-png imagemagick-tiff imagemagick-tools
magick convert /opt/original.png /opt/convert.bpg`
will not work correctly. it will create a file that have BPG extension but is actually a PNG.

please add support for more efficient then JPG web picture standards, like BPG or WEBP.
http://xooyoozoo.github.io/yolo-octo-bugfixes/#swallowtail&jpg=m&bpg=m

they are already supported by imagemagick 7. https://www.imagemagick.org/script/formats.php#supported

mhei, 2018-01-04:
With your Entware hint, I found https://github.com/Entware-ng/entware-oldpackages-ports/blob/master/imagemagick/Makefile

Looks like this library/tool was not yet ported to this new/offical feed, so please file an issue here: https://github.com/Entware-ng/entware-oldpackages-ports
This is why I'm closing this issue here as invalid, simply wrong repository.

Alternatively, the referenced Makefile seems to be a good starting point for re-upstreaming this package. Maybe you find a maintainer or you are interested to maintain this package yourself? Then please create a PR.

closed 2018-01-04 by mhei

eliz82, 2018-01-05:
thanks for the response mhei . i will try to open an issue here https://github.com/Entware-for-kernel-3x/Entware-ng-3x/issues as "entware-oldpackages-ports" have the Issue github option disabled :)

i'm not so good with unix to engage as a package maintainer, sorry.